GoogleCloudPlatform/google-cloud-iot-arduino #5: Add PlatformIO to the project to improve development workflow

alvarowolfx, 2018-04-08:
As I mentioned on the PR #4, I use PlatformIO for embedded environment. I never used for developing a library but I think that we can have a lot of advantages, like declaring the project dependencies from Git or from other Arduino Libraries. This way the current "compile" flow on the `jwt` folder can be changed to use PlatformIO. 

http://docs.platformio.org/en/latest/librarymanager/config.html

Also, we can also how to configure a CI environment to test the library on different boards the project. An example of this can be seen in the PubsubClient project:

https://github.com/knolleary/pubsubclient

What do you think of this? Make sense for this project?

gguuss, 2018-04-09:
Great suggestions, I'll see what can be done for getting checks in place.

gguuss, 2019-03-09:
Just to update the issue here, current plan is to get 1.1.x to a point where there is a "universal" sample that supports all board targets and resolves top issues. After getting 1.1.x to a point that it's stable, we'll migrate to Platform IO on 2.x.x.