GoogleCloudPlatform/google-cloud-iot-arduino #8: Add support for RSA keys

gguuss, 2018-04-12:
We originally were targeting the ATmega328p, which constrained us to Elliptic Curve, with faster MCUs available, it should be feasible to also support RSA-256 keys.

This issue is to capture any progress and requirements for RSA.

alvarowolfx, 2018-04-12:
We can get some ideas from the MongooseOS GCP library. It uses `mbedtls` for the token signature and has support for RSA and ECDSA certificates. But unfortunately, mbedTLS doesn't have support for such low-level devices like the ATMega328p. 

https://github.com/mongoose-os-libs/gcp/blob/99bd8e08e5d9d990482719f9e771dd575d645c21/src/mgos_gcp.c#L108

vinaytumbemane, 2018-05-31:
How to generate Public private key for Arduino Uno ATmega328p. In IOT core documentation only code is given It is not specified where to run the code. I use Windows laptop and when I connect Arduino to my computer only way to communicate with it is Arduino IDE.

Please can any one let me know how to generate JSON Web Token(JWT) signed with private key.

It will be helpful if any one provides link for same.

gguuss, 2018-06-20:
@vinaytumbemane I'm not sure exactly what you're trying to do here but if you're on the Windows laptop, the examples on the [Cloud IoT Core documentation should be helpful](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#device_authentication).

The issue you commented on is for tracking adding support for RSA keys, you may want to create a separate issue for passing credentials via serial.

vinaytumbemane, 2018-06-20:
@gguuss I am using windows system.  I could able to generate RSA key using OpenSSL.

What I am trying to do is, get data(Voltage level and GPS Value) from arduino Uno(I think I should Use ESP8266) and Sensors. And store that data in GCP and Display it on Android app.

I created registry, added device, created pub/sub topic in IOT Core. But I am unable to understand or learn how to send data from arduino/ESP8266 to GCP. There are not many tutorials available on this.

gitandlucsil, 2020-09-02:
Are still working on this? In case of yes, is there any news?

gguuss, 2020-09-02:
@gitandlucsil No updates from me at this time and I am not actively working on this. If you're using an Espressif board, you may want to look at this community tutorial on using ESP-IDF:

https://cloud.google.com/community/tutorials/embedded-c-getting-started