GoogleCloudPlatform/google-cloud-iot-arduino #12: private key str length variance

ptone, 2018-04-19:
[bytes have been changed to protect the innocent]

depending on the keygen, the step to get an encoded private string for MQTT example:

openssl ec -in ec_private.pem -noout -text

generates hex string of a different number of bytes:

const char* private_key_str =
    "63:dc:3c:27:fe:4c:11:c4:a1:21:f2:90:70:b4:42:"
    "e0:02:12:95:25:2a:cf:31:fc:78:95:9f:99:54:4d:"
    "2b:a9";    

const char* private_key_str =   
    "00:ab:64:4a:10:ed:7e:d8:4b:8a:0f:ac:12:94:27:"
    "2d:61:10:02:fc:45:06:8c:ab:bc:07:cd:50:05:ef:"
    "32:a1:15";    

The shorter form works, the longer form does not.

fill_priv_key is at least one of the functions that would need to be modified, not sure about others.

alvarowolfx, 2018-04-19:
Which keygen method did you use? For now indeed, we only have support for the `prime256v1` method and it's the only Ellipse Curve method supported by the Cloud IoT Core platform. You can take a look at the official docs:

https://cloud.google.com/iot/docs/how-tos/credentials/keys#generating_an_es256_key

The other method supported for Cloud IoT Core is the RSA-256, but this library don't have support for it yet.

cirvladimir, 2018-04-19:
Looks like the larger key is prefixed with 00. This is valid as per the spec since the number is the same. However fill_priv_key isn't expecting that so it parses in wrong.
I'll update the function to fix it, thanks for the heads up.

cirvladimir, 2018-04-20:
On second thought made a comment instead asking to delete the extra 00. This also ensures that if someone accidentally makes a key that's the wrong size they'll catch the mistake by reading the comment.

closed 2018-04-20 by cirvladimir