GoogleCloudPlatform/google-cloud-iot-arduino #21: string.h -> String.h conversion in pull_crypto.sh

llanphar, 2018-05-07:
After getting header-not-found compile errors in my dev environment (linux, building for ESP32), I discovered that pull_crypto.sh is changing all includes of string.h in src/crypto to String.h. Switching these back to string.h resolved the problem.

Is there a reason this conversion is being done? My understanding is that the string.h desired here is the standard C string.h with prototypes for memcpy and the like.  All of the instances of this file I have in my Arduino IDE are all lowercase.

gguuss, 2018-05-07:
In Arduino, string means the C/stdlib-string variant wheras String means the Arduino native string type.  Using string in some libraries in lieu of String causes dependency issues. You can convert between the native Arduino type and the stdlib type with `c_str()`.

llanphar, 2018-05-07:
Sorry, I may have done a poor job of explaining that. There are 4 files in src/crypto (ecdsa.cpp, nn.cpp, secp256r1.cpp, sha256.cpp) that, when originally cloned from their repositories in their C form, contain "#include <string.h>". But pull_crypto.sh then does this:
```
# Change string.h to String.h
for f in src/crypto/ecdsa.cpp src/crypto/nn.cpp src/crypto/secp256r1.cpp src/crypto/sha256.cpp
do
  sed -i 's/#include <string.h>/#include <String.h>/' $f
done
```

It's unclear to me what the intent/need for this change is, especially given that the files in question are still straight-up C underneath and still making calls to memcpy(), memset(), etc.  These aren't really string-specific functions, but they are prototyped in string.h.

What is the intended String.h (uppercase) being referenced here? I've not been able to locate any files with that exact name in Arduino IDE, the ESP8266/ESP32 board libraries, or any of the dependencies I'm aware of for this code. 

It's maybe worth pointing out that I'm on linux, so filenames are case sensitive. #include <String.h> (uppercase) won't even compile because that file doesn't exist so far, but reverting to #include <string.h> (lowercase) compiles and runs just fine. 

Did that make more sense? "string" is proving to be a tricky term to use in this context with adequate specificity and clarity:).

gguuss, 2018-05-09:
> Did that make more sense? "string" is proving to be a tricky term to use in this 
> context with adequate specificity and clarity:).

Thank you for clarifying, we are intentionally using the String class that is part of Arduino instead of using the C / C++ string class.  There are [a few discussions online](https://stackoverflow.com/questions/34603609/difference-between-string-object-and-array-of-characters-terminated-with-null) regarding [string vs String](https://forum.arduino.cc/index.php?topic=490250.0), of note, string (null terminated array of characters) is more efficient than the Arduino String class in terms of the memory footprint for two identical (lists of characters).  Internally, the Arduino String class has various conveniences that have some overhead (e.g. dynamic allocation of memory) that are a trade-off for performance but that are also **portable** for devices supporting Arduino.

At any rate, when using some boards (e.g. Genuino/MKR1000/Arduino101) you will encounter issues when using strings with the native libraries. Additionally, in .ino files, the native type for character arrays is the String object.  As such, you must convert types with .c_str() before passing them to libraries that use the `string.h` instead of `String.h`. The spiritual intent of the sketches and library is to provide a Arduino-native developer experience so we opted for Strings as it will make it easier for us to support a broader range of boards with minimal porting. In the future, hopefully less of the details will be apparent to the developer when using our library but for now, much of the board-specific networking exists in the sketches and you do encounter issues when trying to support a broad range of Arduino variants.

Anyways, I'm completely open to using the C string / C++ string types so if you have a compelling reason for using the non-Arduino type, totally open to changing. I just want to make sure that we don't unintentionally make it difficult to port to other Arduino compatible devices. For cases where you want to develop in straight-C or are using a platform other than Arduino, we provide a [C/C++ starting point](https://github.com/GoogleCloudPlatform/cpp-docs-samples/tree/master/iot/mqtt-ciotc) that you can port to your desired embedded environment.

gguuss, 2018-05-09:
Also, as you're not the only person hitting this, I will try some more development from Linux and will see whether I encounter the same issue. Thanks again for pointing out the issue :)

llanphar, 2018-05-09:
I think I see where the disconnect is and how to clear it up. Apologies in advance for covering stuff you might already know; just trying to get us on the same page.

I'm in absolute agreement with you when it comes to using String class vs string class in Arduino, but that isn't the issue I'm raising here; I think you're conflating string (the C++ class) with string.h (the C header file). The intended string.h and its use in these files is completely unrelated to the C++ string class. I propose we disconnect all C++ and Arduino meanings of [Ss]tring from this discussion as they don't play a role and are just adding confusion.

I've always found the naming of string.h to be rather unfortunate because it's only partially about strings and the name presents a red herring in situations like these. There's some good discussion about the nuances of string.h [here](https://stackoverflow.com/questions/9782126/why-memory-functions-such-as-memset-memchr-are-in-string-h-but-not-in-stdli)

That discussion is particularly relevant to this situation because it's the same thing--these crypto files aren't doing anything string-related at all and aren't referencing any of the string-related stuff in string.h. If you comment out `#include <string.h>` on those four files to see what the compiler barfs on, you'll see there are 3 things they need via string.h: NULL, memcpy(), and memset().  memcpy() and memset() are being used to copy and zero out byte arrays, respectively.

I don't have a setup where I can verify this, but I would expect that on machines with case-insensitive file names, there is no difference between `#include <string.h>` and `#include <String.h>` and they'll both resolve to the same string.h file and compile the same binary. But on case-sensitive machines, e.g. linux, the String.h one fails. 

Did that help shed more light on this? Happy to clarify anything.

gguuss, 2018-05-13:
@llanphar Thank you for the continued discussion, your comments have taught me a thing or two for sure! I've looked into [resolving this issue](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/5), and it will be much easier to get validation on PRs, which may require us to wrap/rewrap the string class for PIO.  So I agree, it may be more in the spirit of the JWT library to use the `<string.h>` class and it could potentially make upstream contributions easier while getting us closer to continuous integration on pull requests. I think @cirvladimir should chime in but if you're interested in contributing a fix, I'm happy to review PRs. I'll be looking into CI soon / building on Linux and will be considering the library change when I do.

gguuss, 2018-05-13:
You may want to try the `Arduino.h` header as well, this is resolving the String type for me.

llanphar, 2018-05-14:
Yeah, using `Arduino.h` works for me as well. I was curious, so I took a peek at that file                           and it starts like this:
```
#ifndef Arduino_h
#define Arduino_h

#include <stdlib.h>
#include <stdbool.h>
#include <string.h>
#include <math.h>

...

```
So that's why things compile fine with that. And then further down that same file:
```
#ifdef __cplusplus
#include "WCharacter.h"
#include "WString.h"
...

```
`WString.h` is what defines the `String` class. As for the `string` class, I *think* that's being provided implicitly via `packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/c++/4.8.3/string` (replace path components as appropriate for your setup). But just to clarify, neither of these are involved in the crypto files.

I think it should come down to using either `string.h` or `Arduino.h`. Since these files are coming from elsewhere, my gut tells me it might make most sense to leave it as `#include <string.h>` to minimize the number of local modifications. That should be immune to any String-vs-string-class issues happening elsewhere in the code. But for my purposes, I'd be equally content with `Arduino.h`.

gguuss, 2018-07-06:
I'll consider this resolved - I have removed references to the String class where portable.

closed 2018-07-06 by gguuss