GoogleCloudPlatform/google-cloud-iot-arduino #23: 403 Error

mehmetnyarar, 2018-05-28:
Hi,

We have WeMos D1 R2 mini connected to the Arduino board. Our setup works well with BLYNK app but we would like a customized solution using GC. However, we receive 403 error as follows:

```
{
  "error": {
    "code": 403,
    "message": "The caller does not have permission",
    "status": "PERMISSION_DENIED"
  }
}
```

We followed all the instructions related to devices, pub/sub registrations.

You can see IAM config below:
IAM Members: https://imgur.com/luq0hkF
IAM Roles: https://imgur.com/a/rYD7hjc

I also set certificate expiry to 12 hours just in case. 

For example, should we add a certificate to registries section?
https://imgur.com/CB0uRJ2

We'll appreciate for your help.

sonalidebbarma17, 2018-06-13:
Hello,
Even I am facing the same problem. I also added the certificate but the same error is displayed.
Any help would be appreciated.

decastro-alex, 2018-06-13:
@sonalidebbarma17 @mehmetnyarar same here. Has anyone found a hack around it? :-/

gguuss, 2018-06-13:
Did you:
* Enable the IoT API
* Create the registry on a GCP account with Billing enabled
* Register the EC public certificate with the device registry
* Extract the private key and input it into the device registry

Also:
* Which example are you using
* Which version of the library are you using

sonalidebbarma17, 2018-06-13:
@gguuss 
Hi,
I am using the Basic.ino code and the library version is 1.0.2
I did everything as you have mentioned in the blog but still getting the same result.
Also, I have checked the JWT signature on jwt.io which gave me as signature invalid.

Any suggestion on what can be done next?

Thank You.

gguuss, 2018-06-15:
@sonalidebbarma17 Sorry for the slow responses, I'm on vacation -- this should be working, are you able to connect to your device registry using the virtual device samples?  I have [a demo of how to do this](https://www.youtube.com/watch?v=PSZU-3afKPk#t=13m50s) from a talk last year in India that may be helpful or you can follow [the end-to-end sample](https://cloud.google.com/iot/docs/samples/end-to-end-sample). Once you've verified that is working, we will know it's not a registry configuration issue and could focus on the device side.

gguuss, 2018-06-15:
That the JWT signature is invalid is interesting, you're not seeing your project ID or anything in there?

gguuss, 2018-06-15:
> should we add a certificate to registries section?

Those are used for TLS and are optional, you just need to register the public key for your device. You can see me run through the whole configuration process in [this talk at IO](https://www.youtube.com/watch?v=7kpE44tXQak&t=1876s#t=29m).

sonalidebbarma17, 2018-06-26:
Hi, @gguuss Took a long time to figure out the problem but still no luck. I did successfully connect my virtual device to device registry using the virtual device sample as you have explained in the talk. And yes, I am able to see my project name and the expiration time is also valid in the JWT.
The device I am trying to connect is NodeMCU(esp8266-12e).  I am able to connect this device to Google Cloud IoT Core using mongoose OS, demo-js app and mos tool command but when I am trying with this code I keep getting this error.

sonalidebbarma17, 2018-06-26:
I have figured out the problem. My esp8266 board library was not updated to the lastest version i.e. 2.4.1. When I updated the version, it started working.
@gguuss Thank you. Your blog helped a lot :)

gguuss, 2018-07-06:
Woohoo! I'll close your issue, happy to see you were able to resolve it.

closed 2018-07-06 by gguuss

neuberfran, 2020-09-24:
Hi,
[image]

{
  "error": {
    "code": 403,
    "message": "The caller does not have permission",
    "status": "PERMISSION_DENIED"
  }
}