GoogleCloudPlatform/google-cloud-iot-arduino #25: memcopy not found

pacifickashyap, 2018-06-26:
I am getting error:
Arduino: 1.8.1 (Windows 10), Board: "WeMos D1(Retired), 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Users\kashy\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp: In function 'void NN_Assign(NN_DIGIT*, NN_DIGIT*, NN_UINT)':

C:\Users\kashy\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:85:35: error: 'memcpy' was not declared in this scope

   memcpy(a, b, digits*NN_DIGIT_LEN);

                                   ^

C:\Users\kashy\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp: In function 'void NN_Div(NN_DIGIT*, NN_DIGIT*, NN_DIGIT*, NN_UINT, NN_DIGIT*, NN_UINT)':

C:\Users\kashy\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:248:11: error: 'NULL' was not declared in this scope

   if(a != NULL) {

pacifickashyap, 2018-06-26:
Also getting  fatal error: rBase64.h: No such file or directory
Pleas help to resolve the issue.

closed 2018-06-27 by pacifickashyap

Vishalkpatu, 2018-06-27:
Have you resolved this issue?
I am getting same error. Please let me know if you have solution.
Thanks in advance.

pacifickashyap, 2018-06-28:
Yes,for rBase use : https://github.com/boseji/rBASE64
for memcopy error
#include <cstddef>
#include <string.h> in nn.h and sha256.h