GoogleCloudPlatform/google-cloud-iot-arduino #27: ESP8266 MQTT?

debsahu, 2018-07-03:
This [async MQTT library](https://github.com/marvinroger/async-mqtt-client) supports SSL+TLS. If it possible to use MQTT provided by this library to connect to Google IoT? All I see is the example for MQTT for ESP32.

debsahu, 2018-07-04:
One more issue with Git version, it wont compile until I add `#include <String.h>` in nn.cpp, secp256r1.cpp, sha1.cpp

gguuss, 2018-07-06:
I'd definitely be interested in seeing whether this approach works. For the mqtt_esp32 example that you're encountering the issue with, what version of Arduino are you using and which OS are you building from?

debsahu, 2018-07-06:
I'm on the git version of esp8266-arduino and windows 10 x64 os. I tried adapting this code for esp8266 and failed miserably, with wdt resets with some function in nn.h. don't have time to pursue further, but will come back to it later. Currently comparing all cloud options amazon away/gcp/azure/watson for esp8266.

gguuss, 2018-07-06:
Ahh, for ESP8266, [the ESP8266-http sample](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp8266-http) is the only supported version. I used it in [this demo](https://github.com/GoogleCloudPlatform/iot-hexspider-esp8266) that I [delivered at I/O this year](https://www.youtube.com/watch?v=7kpE44tXQak#t=41m30shttps://www.youtube.com/watch?v=7kpE44tXQak#t=41m30s).  I have been working on porting MQTT to work for the 8266 but there's some strange issue that causes crashes.

debsahu, 2018-07-06:
Thank you for those links, I saw that video that inspired me to start working on this. I can confirm that the http method works for esp8266. Mqtt is still a challenge, if I figure it out I'll keep you in the loop.

As for the esp32-mqtt, I used a doit ESP32 board with latest esp-idf branch on Arduino IDE 1.8.5 Windows 10 x64 machine. Won't compile the git version of this library unless I add string.h to the above mentioned files. The release version complies just fine!

gguuss, 2018-07-07:
> As for the esp32-mqtt, I used a doit ESP32 board with latest esp-idf branch on Arduino IDE 1.8.5 Windows 10 x64 machine. Won't compile the git version of this library unless I add string.h to the above mentioned files. The release version complies just fine!

Oof, I really need to do a maintenance check on the GH version. Been meaning to set up continuous integration so that commits are easier but it's tough and generally [depends on Platform IO](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/5), which is about halfway working.

gguuss, 2018-08-07:
Now that there is SPIFFS support for the CA cert, the ESP8266 may work with MQTT.

manojvellala, 2018-11-05:
> Ahh, for ESP8266, [the ESP8266-http sample](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp8266-http) is the only supported version. I used it in [this [...]

Solution: 
1) Disable the CPU watchdog before any crypto function and enable it afterwards.
2) the call back functions are not set correctly in the code - The callback function pointer is Null. Please trace the call back functions in the Mqtt code. This cause another crash.

debsahu, 2018-11-18:
@gguuss on ESP8266, I was able to use BearSSL along with PubSubClient to communicate with secure MQTT servers here: https://github.com/debsahu/ESP_MQTT_Secure Can we do something similar here?

gguuss, 2018-11-18:
@debsahu This looks promising, maybe another example using BearSSL and Esp8266 could help.

debsahu, 2018-11-19:
@gguuss check these out, https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples bunch of BearSSL examples. See example,  BearSSL_Validation.

gguuss, 2018-11-20:
Example included [in release v1.0.5](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/releases/tag/v1.0.5)! Hopefully we'll have a more integrated version in a future release.

closed 2018-11-20 by gguuss