GoogleCloudPlatform/google-cloud-iot-arduino #30: Corrupted memory

jeanmatthieud, 2018-07-04:
I tried to use the MQTT ESP32 sample and simplify it.

A call to `mqttClient.setCallback(callback);` in the `setup()` should be enough.
By logging the address of `callback` in `PubSubClient.cpp`, I realized that the address changed between the `PubSubClient::setCallback()` and the `PubSubClient::loop()`.
The example resets the callback pointer just before subscribing to the config topic.
If you comment that, the example will crash when calling the callback.

I lost a lot of time before understanding that.
I do not know where the memory get corrupt.

I have other projects using PubSubClient and a Mosquitto server without any issue.
I think it's related to the IoT Core layer.

jeanmatthieud, 2018-07-04:
After more digging, it's exactly the same for the know issue about the line 266 referenced in the readme.

I avoid it by resetting the stream pointer, as it is done for the callback, just before subscribing:
```
mqttClient.setCallback(mqttCallback); // Corrupted memory ?
Stream *ptrNullStream = NULL;
mqttClient.setStream(*ptrNullStream); // Corrupted memory ?
mqttClient.subscribe(configTopic.c_str(), 1);
```

gguuss, 2018-07-06:
This is super helpful, if you want to commit a PR with a fix for the issue and have signed our CLA, I'm happy to review and merge. Otherwise, I'll see if I can repro and fix.

jeanmatthieud, 2018-07-08:
I will be happy to commit a PR, but I don't where the memory becomes corrupted.
I tried to change your example to run it on the ESP8266, and this memory corruption cannot be worked around with my previous comment, as the PubSubClient callback declaration is not the same:
```cpp
#ifdef ESP8266
#include <functional>
#define MQTT_CALLBACK_SIGNATURE std::function<void(char*, uint8_t*, unsigned int)> callback
#else
#define MQTT_CALLBACK_SIGNATURE void (*callback)(char*, uint8_t*, unsigned int)
#endif
```

If this issue is fixed, the MQTT example can be ported to ESP8266!

gguuss, 2018-08-02:
There was potentially a memory leak in the int_to_string function that I modified in some recent changes to the library. Let me know if it helps with where you've been stuck.

gguuss, 2018-11-18:
@jeanmatthieud The fix with the stream pointer was very helpful in getting the callback to stop crashing. [This has been incorporated in this PR](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/54).

closed 2018-11-18 by gguuss

jeanmatthieud, 2018-11-18:
Great I will try that next week !