GoogleCloudPlatform/google-cloud-iot-arduino #36: Create CloudIoTCoreHTTPClient in library

gguuss, 2018-08-07:
In CloudIoTCoreDevice we should add:

* A common exponential backoff function based on ESP8266 / ESP32 HTTP examples
* Creates a common set of headers for GetConfig, SetState, and SendTelemetry

We should then create an HTTP equivalent to the MQTT client type that:

* Manages lifecycle of the HTTP connection
* Can be constructed from our supported device types

closed 2019-02-09 by gguuss

alexodus, 2019-04-18:
Why this isssue/enhancement was closed?!

gguuss, 2019-04-25:
Hey, sorry for any confusion, as of 1.1.x the library has changed to be simpler and only use MQTT.

gguuss, 2019-04-25:
If we wanted to move forward with these enhancements in the HTTP-supported version of the library, the way to go would be to fork [the 1.0.6 release](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/releases/tag/v1.0.6).