GoogleCloudPlatform/google-cloud-iot-arduino #37: hi..I want to know is it possible to store data in firebase and have the sub / pub in the same function using the iot core in google cloud? any suggestion?

hanadjerbi1, 2018-08-16:


hanadjerbi1, 2018-08-17:
okay...my first question
-how you make a connection between google cloud and firebase?

hanadjerbi1, 2018-08-17:
okay...my first question
-how did you make the connection between google cloud and firebase?

hanadjerbi1, 2018-08-28:
>     I have to posts that explain this kind o integration with Cloud IoT Core @hanadjerbi1 https://github.com/hanadjerbi1 . I recommend you to take a look at them and if you have more questions, feel free to contact us.
>
>     https://medium.com/google-cloud/build-a-weather-station-using-google-cloud-iot-core-and-mongooseos-7a78b69822c5
>
>     https://medium.com/google-cloud/gps-cellular-asset-tracking-using-google-cloud-iot-core-firestore-and-mongooseos-4dd74921f582

where i could run this command"npm install -g firebase-tools" with cmd ?

gguuss, 2018-11-13:
@hanadjerbi1 I'm going to close this issue as Firebase integrations are external to this project.

closed 2018-11-13 by gguuss