GoogleCloudPlatform/google-cloud-iot-arduino #38: refreshing jwt doesn't work

alantung525, 2018-08-26:
Hi, 
I use heltek wifi kit 32 device and Esp32-ssd1306 example to connect to google iot. 
Following the steps,  it's good for me to  connect to google cloud. 
Unfortunately, I get trouble when I like to refresh jwt. 
After many trouble shooting, I find a question that the jwt string produced from createJWT function is invalid signature at  https://jwt.io(The header and payload look good from the jwt string).
I try to use the signature verified jwt string which is produced from https://jwt.io to connect to google iot and refresh jwt, it does work fine. 
Does the invalid signature jwt cause the refresh failed? or Do I have something wrongs?
Could you please give me a hand, thank you very much.

gguuss, 2018-09-06:
It could make sense to detect failures and try to refresh the token instead of assuming the token will be valid until its expiry.  Thanks for pointing out the issue!

alantung525, 2018-09-13:
I find a difference between your base64 char. definition and base64url char. definition. The base64url uses "-_" at the last two char.. After using the "-_" instead  of "+/",  the signature of output jwt is verified. Unfortunately,  I still got signature invalid when I reproduce jwt after time expiration. I don't know what's going on. Do you have any idea for this kind of situation. Thanks.