GoogleCloudPlatform/google-cloud-iot-arduino #39: PubSubClient Issues (and fixes)

cyn42, 2018-08-31:
Couple of comments from someone who has just recently been able to get this client to work with an ESP32. Perhaps a README update might be helpful for others?

I ran into 2 issues which were resolved by PubSubClient fixes...

1. In the README for Google Cloud IoT JWT, it says to comment out line 266 of the PubSubClient.cpp. That line is now actually line 272; might be better to say which code needs to be commented out

2. I was receiving a very frustrating error in my device info in IOT Core - "mqtt: SERVER: The connection was closed because MQTT keep-alive check failed.". I updated the MQTT_KEEPALIVE in PubSubClient.h, however that didn't help. It turns out the non-intuitive root cause was that my MQTT payload was too big... updating the MQTT_MAX_PACKET_SIZE from 128 to 256 solved the problem.

gguuss, 2018-09-06:
Thank you for the errata.  I'll try and update the docs and library soon.

gguuss, 2018-11-18:
This should be fixed now, I'll be releasing a new version soon.

closed 2018-11-18 by gguuss