GoogleCloudPlatform/google-cloud-iot-arduino #40: Write to specific Pub/Sub

Gior80, 2018-09-05:
Hello,  I'm able to connect to library and everything work smoothly.

On google not core  I setup specific state topic related to pub/sub

projects/registry1/topics/temp | temp
projects/registry/topics/ledstate | led

Now,  I'm wondering how to write and get notified about those specific topics.

gguuss, 2018-09-06:
The documentation on [Publishing to separate Pub/Sub topics](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#publishing_telemetry_events_to_separate_pubsub_topics) may be helpful here. You would need separate subscriptions to each topic to separate the messages published to each topic.

Gior80, 2018-10-01:
Yes thanks !!

closed 2018-10-01 by gguuss