GoogleCloudPlatform/google-cloud-iot-arduino #42: How to publishTelemetry/State with MKR1000?

Razzo1987, 2018-09-16:
Hi,
I have seen the examples/Basic-MKR1000/ but there isn't any publish data.
I have tried to use the CloudIoTCoreMQTTClient on MKR1000 but the class use WiFiClientSecure and the code doesn't compile.

Someone have an example to publishTelemetry/State with MKR1000?

thanks

gguuss, 2018-11-18:
@Razzo1987 Did you install the root certificate as described [here](https://www.hackster.io/arichetta/add-ssl-certificates-to-mkr1000-93c89d)?  Also, are you seeing the configuration messages?

gguuss, 2018-12-06:
Closing issue, this is now supported in the lwMQTT examples.

closed 2018-12-06 by gguuss

ripilot, 2019-04-02:
Hi all, whats the significance of 1510644967 in the code above?

gguuss, 2019-04-03:
If you convert seconds since epoch of 1510644967, it's a date around November of 2017, roughly the first time of publish for the sample.  I believe you could use any value greater than zero, but I just wanted to make sure the current time stamp coming from the WiFi library was valid and recent.

ajmalvb78, 2021-03-09:
hii all, 
iam sending data to a gcp cloud function successfully from my aurdino mkr wifi 1010 , but when my friend tries from another location( netherland ) he was not able to connect to iot core.my location was india . he and i was sending the data to same device , and using the same program , same private key, and added certicate for "mqtt.googleapis.com" on both of the aurdino mkr devices, does someone have any idea about the changes needed?
thanks in advance

raphael-bmec-co, 2021-03-09:
Are you trying to do this at the same time? Google IoT core will only accept one connection from a device at a time. And if the connection is not closed or has not timed out then it will not accept a new connection. Try making a different device with it's own key pair.