GoogleCloudPlatform/google-cloud-iot-arduino #45: Publish state

Gior80, 2018-10-01:
I could setup a working example with publish telemetry but I still cannot make the publish state work . 
Actually , for test , I used same payload of telemetry and, on server side , I used the same “telemetry topic” for the state . Looking on iot core interface look like device never sent an information state , the code on device look running well . Am I missing something ?

gguuss, 2018-10-01:
I have spent less time using PubSub topics for validating state.  If you go to the [Cloud IoT Console](https://console.cloud.google.com/iot) you can find the last couple of state changes listed there.

Gior80, 2018-10-04:
also here:   config is sent ,   device send a state but looks like is not related to the config received.  Is there a special payload to send to ack configuration?
[image]

Gior80, 2018-10-04:
I got to the point .

Acknowledge of conf is doin by sending back a puback message .

Totally in fog :   how to do that ???

gguuss, 2018-10-04:
I believe that you want to use QoS 1 if you want to ensure at least once delivery of the messages (configurations) published to the config topic. For using configuration and state, a configuration message could be something like `{fan1: 500}` to set the fan speed to 500 rpm. At this point, it makes sense to report back the RPM speed of the fan back to Cloud IoT core using state.  In this way you can verify if a device is unable to respond to a configuration change.

gguuss, 2018-11-18:
Closing, as I'm not sure this is an issue as much as support for the library and this was discussed in separate issues.

closed 2018-11-18 by gguuss