GoogleCloudPlatform/google-cloud-iot-arduino #46: Board-independent MQTT to Google Cloud IoT

demison, 2018-10-07:
Trying to understand how to use Libraries with another board (for ex. Arduino Uno + Ethernet Shield).
Is that possible to do myself with not lots of knowledge?
But, at first, I just can't compile any example because of header issue: cstring in some h-files.
Compiling with Arduino IDE 1.8.7 on Mac OS.

gguuss, 2018-10-23:
The master library is still in development, you may want to try [one of the releases](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/releases) instead of the master branch for now. In terms of board-independent MQTT support, we don't have a good solution yet. If your ethernet shield can do SSL, [the MKR-1000 example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Basic-MKR1000) uses an Arduino-native library that should be similar.

gguuss, 2018-11-20:
The library is designed to make it easy to use get configuration from the Cloud IoT Core Device object, if you look at ESP8266-lwmqtt, you can see a basic example for integrating without using the encapsulated device object.

gguuss, 2019-02-09:
I'm currently planning on seeing if I can make a merged sample for the lwmqtt samples that works broadly across Arduino boards.

gguuss, 2019-04-05:
First attempt at this is merged in [this CL](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/100).

szundi, 2019-11-16:
nice job!

gguuss, 2020-01-17:
Closing this is now in master.

closed 2020-01-17 by gguuss