GoogleCloudPlatform/google-cloud-iot-arduino #47: Jwt  Refresh and Timing

Gior80, 2018-10-09:
Hi Gus, 

working a lot on this library, you made a really nice job! 
I'm actually facing of other pair of points I could not understand: 
- JWT refresh: Actually IoT Core need need to refresh key maximum every 24 hours. looking at code seems this is actually not covered I'm saying wrong ? 
- Timing.  I'm trying (successfully ) to set device config via cloud functions. it works but has a really hi delay ( about 9 sec).  would u suggest put my attention on firmware side or this kind of delays are normal into Matt enviroments?

Gior80, 2018-10-09:
[image]

seems it die after 1300 seconds.

Gior80, 2018-10-10:
UPDATE:  I got another small  improvement can be made:

KEEPALIVE FAILING: IOT CORE Documentation report an optimal value is 60 Seconds for the keepalive.  Actually it's not possible to set this timing directly from code.  It comes from the pubsubclient library used for mqtt: so in pubsubclient.h we just need to set the keepalive to 60 sec. 

regarding instead the expiration token: 
Here in Europe the timing is around 20 min ( 1399sec) after JWT token expires.
could we try implement a solution like guy here : 
SEE LINE  96>
https://github.com/teifip/gcic-mqtt-client/blob/master/index.js

I'm really stuck on this JWT  refresh and I can't solve it!

thiagolc3, 2018-10-17:
Hi Gus and Gior80. I'm having the same problem, after 1 hour JWT expires. I tried to find in the code, where is JWT updated? I got the impression of being in "void loop ():"

if (counter <1000)
    counter ++;
...
else {
       counter = 0;
       // would it be here?
     }

Thanks for attention.

Gior80, 2018-10-17:
I think timing is depending from server . 20min for Europe and 1hour us side . 

I have same impression too. Only way to manage that is to add method in class to refresh (or stopping and reconnecting the client). Gus able to help you working on this if you can teach me right approach to solution as i can’t find lot of documentation

gguuss, 2018-10-18:
@thiagolc3 @Gior80  - I believe the `getJwt` function in each example is intended for refreshing the token. Here's an example from the [ESP8266 HTTP example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp8266-http/esp8266_wifi.h#L47-L56).  Been really busy lately but definitely hear y'all that the library needs some maintenance.

Gior80, 2018-10-18:
I saw that cycle , Mmmmm that’s my fault then ?

I’m using esp32 so I think it never enter in this cycle then ??? Gonna check !!!!!

thiagolc3, 2018-10-19:
Hello @Gior80 and @gguuss. I put getJwt function in void loop(). The function generate a new token, but it expires anyway.
I put a ESP.reset() in mqttConnect, and that's how it worked.

Gior80, 2018-10-22:
hi,  played a little with code......

in CLIENT:

String CloudIoTCoreMQTTClient::getJWT() {
  if (iss == 0 || time(nullptr) - iss > 3600) {
    iss = time(nullptr);
    jwt = device.createJWT(iss);
  }
  return jwt;
}

looks like above code is intended for refreshing JWT. btw looking at function we call in device:

String CloudIoTCoreDevice::createJWT(long long int current_time) {
  jwt = CreateJwt(project_id, current_time, priv_key);
  return jwt;
}

this just create the updated JWT.    so what is unclear : 
how to communicate to google we updated jwt ?   Would I need to re-create device and client + reconnect.  I'm probably missing something here, but lack of documentation made everything really unclear.

gguuss, 2018-10-23:
Thanks for your investigation so far @Gior80 - in other languages that use MQTT clients based on the PAHO client, the client does need to be destroyed and then recreated to build new connection credentials, which is most likely the source of failure here.  There is an [example on the PubSub client of reconnect](https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_reconnect_nonblocking/mqtt_reconnect_nonblocking.ino#L29-L37), I'm guessing you will also need to [call disconnect](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L129) if you're explicitly trying to force regeneration of credentials.

Gior80, 2018-10-23:
Hi @gguuss that could be certainly part of problem, btw looks like I can change the jwt now using the createJWT call from loop.  if is in the certain timeframe seems to  work. every 1000 seconds jwt is updated and uptime reset.

{"Uptime":"999","Sig":"-46","lv_1":"0 0","ChargeState":"","Chrgcount":"0","IsRent":"YES"}
Publish message
{"Uptime":"1000","Sig":"-47","lv_1":"0 0","ChargeState":"","Chrgcount":"0","IsRent":"YES"}
Publish message
{"Uptime":"1","Sig":"-47","lv_1":"0 0","ChargeState":"","Chrgcount":"0","IsRent":"YES"}
Publish message
{"Uptime":"2","Sig":"-48","lv_1":"0 0","ChargeState":"","Chrgcount":"0","IsRent":"YES"}

btw I'm still facing the disconnection error, and I go deeply in that as this is driving me crazy.   I'm. using  mqtt.googleapis.com as server and I'm facing this error exactly every 1399 second.    When client disconnect I'm getting reconnection error from pubsub:

failed, status code =-2 try again in 5 seconds

other people facing same issue with pubsubclient , resulting this mean server could not be reached.....  STRANGE!!!! 

so I started logging ip ping to Mqtt.googleapis.com. What I noticed so far is that ping resolve to 2 different ip address dynamically ( load balancing ?????).
74.125.133.206
173.194.76.206

Looks to me ( just an idea at this stage) that something strange happening here,  like sort of caching is happening and make the device point to nothing. 

First things I try to do is making the client pointing to a fixed IP address from client , btw seems not work.  As you are internal Google , could u confirm me ip switching is happening ? 

Giorgio

Gior80, 2018-10-23:
I attach the -2 error meaning 

> The -2 means the underlying WiFiClient could not establish a tcp connection to the broker. There's nothing in PubSubClient that affects this. This is why I suggest you test with one of the other ESP8266 examples that don't involve this library to rule out anything to do with the basic setup.

gguuss, 2018-10-23:
I found that unless I destroy and recreate the PubSubClient, the client will be disconnected.  Repro is in my [refreshjwt branch](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/refreshjwt).

Gior80, 2018-10-23:
I used a different approach,   in client class I'm basically calling reconnect without calling PubSub disconnect. pushing it every 60 seconds
SEEMS..  I repeat seems , to work.   it is connected from 40 min without any problem. keep u updated

String CloudIoTCoreMQTTClient::getJWT() {
    Serial.println("jwtcheck");
  if (iss == 0 || time(nullptr) - iss > 60) {
    iss = time(nullptr);
    jwt = device.createJWT(iss);
      Serial.println("MQTT connecting ...");
      String pass = this->getJWT();
      Serial.println(pass.c_str());
      const char *user = "unused";
      String clientId = device.getClientId();
      Serial.println(clientId.c_str());
      if (mqttClient.connect(clientId.c_str(), user, pass.c_str())) {
          Serial.println("connected");
          if (configCallback != NULL) {
              String configTopic = device.getConfigTopic();
              Serial.println(configTopic.c_str());
              mqttClient.setCallback(callback);
              mqttClient.subscribe(configTopic.c_str(), 0);
          }
      } else {
          Serial.print("failed, status code =");
          Serial.print(mqttClient.state());
          Serial.println(" try again in 5 seconds");
          /* Wait 5 seconds before retrying */
          delay(5000);
      }
      
      
  }
  return jwt;
}

Gior80, 2018-10-25:
hi gus, 
thanks for your work.  I relook @ code and documentation and I finally arrive to an acceptable solution.  basically now google accept the 24hours value for the jwt .   by putting right value in jwt.cpp for exp I think that we can got this reconnection ( loose time ) every 24 hours. 

String payload = "{\"iat\":" + int_to_string(time) +
                   ",\"exp\":" + int_to_string(time +  **SET EXP VALUE IN SECONDS**) + ",\"aud\":\"" +

if you need help or test in your work just let me know how can hep and will be a pleasure.

meantime 

THANKS

Gior80, 2018-10-26:
@gguuss  

you will hate me,  I know....

I tested solution ad now the above jwt problem is resolved, btw we have a new one.
I leave sensor overnight for testing and at some certain point connection fail. The console reports error about keepaliave fails .  I then raised up the keep alive time to 60 seconds. 
no way....

it keep fails with related  status code rc=2  :  can't create the network connection
if I reset the board the connection start again perfectly. 
looking at this guy with same problem:

https://github.com/knolleary/pubsubclient/issues/185

he mention in end the issue is IP related and let me think as my first assumption of the IP address change for load balancing.  

in end, have u been able to get a stable connection over time ?

Gior80, 2018-10-30:
hi @gguuss  I remade your software using native Paho client and it's now working like charme.

think u are right and most of problem derive from PubSub  client.

gguuss, 2018-10-30:
If you have a fix and can submit a Pull Request or demo in working branch, I'll happily take a look.

Gior80, 2018-10-31:
@gguuss I  asked permission and can't publish code, btw if you give me email I can send u some working fuctions.

gguuss, 2018-10-31:
@Gior80 No problem, please add a link on this issue with the updated Paho MQTT client you're using and any helpful code examples demonstrating resolutions to the client usage.

Gior80, 2018-10-31:
@gguuss   I simply think client as native is more stable and has more many option can be set...  
btw:

- here the lib https://github.com/256dpi/lwmqtt (they also have Arduino like with tls examples) 

- i simply got   JWT  lib from your code
- I setup recreated the methods for construction of the various clientId and topics 
- I setup the MQTT client as here: 
- I modified the exp parameter in jwt to have a 6 hours interval 

https://github.com/256dpi/arduino-mqtt

I used exactly same connection option as here, look on MQTTCLIENT_PERSISTENCE_NONE and clean session.  if you need helps let me know and I'm available.

https://cloud.google.com/iot/docs/how-tos/mqtt-bridge

 MQTTClient_create(&client, opts.address, opts.clientid,
 MQTTCLIENT_PERSISTENCE_NONE, NULL);
  conn_opts.keepAliveInterval = 60;
  conn_opts.cleansession = 1;
  conn_opts.username = kUsername;
  conn_opts.password = CreateJwt(opts.ecpath, opts.projectid);
  MQTTClient_SSLOptions sslopts = MQTTClient_SSLOptions_initializer;

one question from my side:   how you generated the SSL cert?  using the command you suggest I'm getting : 

TLS session ticket lifetime hint: 100800 (seconds)
    TLS session ticket:
    0000 - 00 67 1a 0e d7 b7 56 b8-61 23 0a 69 5b bc c0 25   .g....V.a#.i[..%
    0010 - 17 62 e2 44 aa 09 f9 4e-f2 8e 81 9f e5 60 6c 02   .b.D...N.....`l.
    0020 - 7f bc 1f 3b 29 86 9a 38-ad b0 a5 29 16 87 2e 1a   ...;)..8...)....
    0030 - ce 9e 79 b0 0e 50 16 61-03 60 0e 1b 27 f1 f7 c1   ..y..P.a.`..'...
    0040 - ba 65 82 e8 e4 10 72 dd-2d ab 0a 59 cb 40 d7 6b   .e....r.-..Y.@.k
    0050 - d8 bd af dd 72 79 de 02-f4 67 fd e7 a1 ec c6 3a   ....ry...g.....:
    0060 - d1 4c b2 85 b0 6a 61 d7-88 de 51 6f 2f 88 0f 85   .L...ja...Qo/...
    0070 - 2d ce db 1a 8c 0b 2e 8e-3c 1b e0 4b 1f 75 3e 90   -.......<..K.u>.
    0080 - d1 64 a0 c1 eb 0c 18 20-1f 7f 59 2d 45 06 c0 bd   .d..... ..Y-E...
    0090 - 20 61 00 34 9a 9e fb 11-56 6d 85 ba 23 cd cb 65    a.4....Vm..#..e
    00a0 - 9c c6 1f 29 62 fb be 80-34 69 67 6d 34 96 eb 6c   ...)b...4igm4..l
    00b0 - 49 87 4b 2b 13 a9 f3 cd-4f 4c 6d 05 1d 60 25 e4   I.K+....OLm..`%.
    00c0 - 46 f0 68 c8 df cb a6 56-d3 28 fc db 64 d6 31 24   F.h....V.(..d.1$
    00d0 - c4 c1 a4 53 b3                                    ...S.

    Start Time: 1541001405
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)

Gior80, 2018-11-05:
We can consider close.   Gus if leave me an email I can send u codes. 

Giorgio

gguuss, 2018-11-05:
@Gior80 Feel free to email me `github at gusclass.com`

closed 2018-11-05 by gguuss