GoogleCloudPlatform/google-cloud-iot-arduino #52: In Http example: fingerprint does not work, instead use RS/EC key in Hex format

manojvellala, 2018-11-05:
Fingerprints are deprecated. 
The Following code in the Http example's config has to be changed.
const char* fingerprint =
    "7C:D4:99:11:FE:FC:0D:78:C0:A9:C1:18:52:1D:3F:0B:8B:38:C9:90";

Solution: Instead use RSA/EC key in Binary format.

Use "xxd" command on der format of the public key to obtain the Hex format of the key aling with its length. :  https://www.poftut.com/use-linux-xxd-command-tutorial-hex-binary-operations-example/

gguuss, 2018-11-13:
For ESP8266, I have [found a way to install certificates on the device in Flash memory](https://hackaday.io/project/12482-garage-door-opener/log/45617-connecting-the-esp8266-with-tls) for the root cert. I'm currently in the middle of fixing a few other issues but hopefully using the filesystem resolves the issue you're encountering (an expired cert fingerprint).

gguuss, 2018-11-18:
We could remove the server fingerprint from the examples, as of 1.0.3-alpha, you can install the cert into device flash using the upload tool.

gguuss, 2018-11-21:
Closing, now have examples of binary cert and SPIFFs cert storage.

closed 2018-11-21 by gguuss