GoogleCloudPlatform/google-cloud-iot-arduino #53: ESP32 MQTT-GCP connectivity Example code:setCAcert()

harishudhay, 2018-11-05:
1) The key has to be in hex format
2) It has to be typecasted to unit_8* format
3) The key size has to be passed as second parameter.

Solution : use the "xxd" command on DER format of the public key.

gguuss, 2018-11-27:
@harishudhay Can you share the full commands to retrieve the cert and then produce the binary formatted certificate?  I'm having trouble going from this:

    wget pki.goog/root.pem

to the DER format:

    openssl x509 -in roots.pem -outform der -out cert.der

and then getting the key as uint_8*, when I call:

    xxd cert.der

I'm not seeing the bytes I expect.

gguuss, 2019-02-28:
This is now added to the [Notes on the certificate](https://github.com/googlecloudplatform/google-cloud-iot-arduino#notes-on-the-certificate) section of the README.

closed 2019-02-28 by gguuss