GoogleCloudPlatform/google-cloud-iot-arduino #56: Error, client state: -1

alexodus, 2018-11-19:
in the esp32-mqtt example I obtain always this result. I'm using platformio with espressif 32 v1.5.0.

Gior80, 2018-11-19:
Your buffer size is too short

alexodus, 2018-11-19:
the same problem with 2048 also! :(

alexodus, 2018-11-21:
1024 2048 for MQTT_MAX_PACKET_SIZE produce the same error (also directly into the CloudIoTCoreMQTTClient.h file and print the value on serial in order to verify that it's ok) .

gguuss, 2018-11-21:
This could also be caused by incorrect configuration. Maybe try eliminating one of the variables and use one of the HTTP examples; if it doesn't work with HTTP, it definitely won't work with the MQTT. Also, if you're using the library from before 1.0.4, there was a bug where the JWT was not correctly getting generated, try updating the library too.

alexodus, 2018-11-22:
I used the last versione (1.0.5) for my test with 1024 and 2048 buffer size. I'm working on ESP32 and there isn't http example for it. I'm in wrong?

Gior80, 2018-11-22:
Alessandro,

Try to change size directly on pubsubclient configuration file . Btw also look at mqtt error codes , you can find a good tables of it on lwmqtt arduino repository

alexodus, 2018-11-22:
> Alessandro,
> 
> Try to change size directly on pubsubclient configuration file .

WOW! Now with 1024 (with 2048 esp32 go in bootloop) on file pubsubclient.h I have this:
`Error, client state: 5`

A big step forward! :P

gguuss, 2018-11-22:
@alexodus the esp32-ssd1306 example is HTTP, probably could use a rename ^_^.

alexodus, 2018-11-23:
SOLVED! I had only set MQTT_MAX_PACKET_SIZE more than 324 into pubsubclient.h (I don't like it but I think this is a platformio problem). MQTT_CONNECT_UNAUTHORIZED happen because I modified the private key for test :P

alexodus, 2018-11-23:
Best solution for platformio: add this line on platformio.ini file
`build_flags = -DMQTT_MAX_PACKET_SIZE=384
`
see [this](https://github.com/knolleary/pubsubclient/issues/110) issue on pubsubclient library

gguuss, 2018-11-23:
Ahh, thank you @alexodus may be worth adding to the readme, if you want me to do it, happy to accept an issue, if you want to make a PR, happy to review.

gguuss, 2018-11-27:
Added note to the readme, thanks for the errata!

closed 2018-11-27 by gguuss