GoogleCloudPlatform/google-cloud-iot-arduino #58: Error, client state: -2

johanso-au, 2018-11-21:
Getting an "Error, client state: -2" error. I'm unsure where to start looking for the root cause, as the CloudIoTCoreMQTTClient::mqttConnect() function doesn't make it clear where 'state' is set.

I think it may be related to a JWT auth issue, as when I turn on debug, get the generated JWT, and run it through a debugger (https://jwt.io/#debugger) the signature is invalid with my private key.

Is there any more verbose debug logging that can be used to find out what the cause of an abnormal client state could be?

gguuss, 2018-11-21:
All of the errors in CloudIoTCoreMQTTClient are coming from the PubSub MQTT client that it wraps. Check [the State conditions for its low-level message explanations](https://pubsubclient.knolleary.net/api.html#state).  You may want to try one of the HTTP samples to get better information about what's going on but based on the docs, you should turn on debug logging and make sure that your client ID is correctly getting specified.

johanso-au, 2018-11-22:
Thanks Gus. -2 is a connection failed.

Is it worth adding that State error table to the readme?

gguuss, 2018-11-22:
That's a good idea, a link also could be included in the debug log.

gguuss, 2018-11-23:
We may also move to lwMQTT, the examples in the master branch feel a bit easier to follow than the wrapper.

closed 2018-11-29 by gguuss