GoogleCloudPlatform/google-cloud-iot-arduino #62: Not leaving the while loop (connecting)

rscarlat, 2018-11-28:
I am trying the MKR1000 lwMQTT example.
I created all the everything in the GCP following the commands in README.md, and added the correct values in ciotc_config.h.
The board connects to wifi, is refreshing the JWT, but doesn't exit the while loop after the "connecting" message.
The loop returns "0" for both, the lastError and returnCode.
I also printed the returned jwt before the lastError and returnCode, and it looks correct.
There are no logs in the GCP's Stackdrive.
I don't understand why the device doesn't connect.

rscarlat, 2018-11-29:
Got it.
I managed to connect.
I missed the step you recommended in the end of the ciotc_config.h: 

// TODO(you): Install root certificate to verify tls connection as described
// in https://www.hackster.io/arichetta/add-ssl-certificates-to-mkr1000-93c89d

I didn't perform it, because I didn't know for what server I need the certificate. I found the server after you posted the issue #61: mqtt.googleapis.com.
After updating the firmware with the missing certificate, it connects.
Probably this needs to be emphasized in the instructions. Thank you.

gguuss, 2018-11-29:
Happy you found the solution, I'll think about ways we can make the instruction more clear 👍

closed 2018-11-29 by rscarlat