GoogleCloudPlatform/google-cloud-iot-arduino #64: publishTelemetry with binary data

alexodus, 2018-12-01:
Please support publishTelemetry with binary data (aka const uint8_t* payload, unsigned int plength, like PubSub library)

gguuss, 2018-12-04:
What happens when you Base64 encode the data? Is this specific to the lwMQTT examples?

alexodus, 2018-12-04:
I want to send a large amount of data and I want stay under the 1024 byte limit. Base64 is not ok.

gguuss, 2018-12-06:
Got it, makes total sense.  Would it be helpful to add another MQTT example that shows this? (I've been trying to come up with an excuse to learn i2s with the esp32.

gguuss, 2019-03-26:
An update here, in the existing samples you have access to the raw MQTT device, which can accept a byte array:

```c++
  if (millis() - lastMillis > 60000) {
    lastMillis = millis();
    mqttClient->publish("/devices/arduino-taste/events", data, 255);
  }
```

It could make sense to expose this as a helper.

gguuss, 2019-04-24:
This has been added in the release v1.1.5.

closed 2019-04-24 by gguuss