GoogleCloudPlatform/google-cloud-iot-arduino #65: API support for multiple topics

alexodus, 2018-12-01:
Since February 21, 2018 (https://cloud.google.com/iot/docs/release-notes) "device registries can now have multiple associated Cloud Pub/Sub topics". Your API support only one topic. I'm in wrong?

https://cloud.google.com/iot/docs/how-tos/devices#creating_a_device_registry_with_multiple_pubsub_topics

gguuss, 2018-12-04:
You could tailor the lwMQTT samples to publish to multiple topics pretty easily. The CloudIoTMQTTDevice wrapper on the other hand... probably could use a refactor.

Gior80, 2018-12-04:
@gguuss does lwmqtt implements jwt renewal ?

gguuss, 2018-12-04:
I tested it on the ESP8266 and when the client reconnects a new JWT and connection credential are created.  I'm not 100% sure the device didn't reset and have not closely assessed the refresh flow.

gguuss, 2018-12-04:
JWT refresh should happen in [getJWT](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h#L44-L52) when [it's called by mqttConnect](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h#L96-L99).

gguuss, 2018-12-04:
I believe all of this functionality should be encapsulated in CloudIoTMQTTJwt but I want to get the examples with lwMQTT into a good state and then have a major version bump and fully migrate to PlatformIO before trying to do this.

Gior80, 2018-12-21:
yep , that would be great have this parameter ( Jwt expires ). included into the object creation ,  if can help let me know ! 

Giorgio

gguuss, 2019-02-09:
As MQTTDevice has been refactored, it could make sense to move this to `ciotc_config.h` as well and just configure with an array of topics. We should also add the option for commands and gateway topics.

gguuss, 2019-03-22:
An update, release 1.1.3 will have support for subfolders when publishing telemetry.

gguuss, 2019-03-26:
@Gior80 JWT refresh should now work, closing this issue as I added support for publishing to separate subfolders in the 1.1.3 release.

Next I'll be making one unified sample that supports all hardware from a single example.

closed 2019-03-26 by gguuss