GoogleCloudPlatform/google-cloud-iot-arduino #68: esp32 advanced

Gior80, 2018-12-21:
Hi Man :) 

tested the new esp32 advanced and got this strange behaviour.  I'm using exactly same previous ctcio_config with working key.  Seems to connect then die then reconnect and so on ... 

I have not touched code except for CTCIO .   saw u put a waiting new waiting method in code, btw any way to have more info on debug ?  

Publish message
Connecting to mqtt.googleapis.com
MQTT connecting ...
JWT now:
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1NDU0MTExNjAsImV4cCI6MTU0NTQxNDc2MCwiYXVkIjoiY2hhcmdlc3BsIn0=.m6ZyoAWPmyumJZB6ZsXmKcpW1HgPZSsx/r7+8OHpXImXoy9A+Ww/RLkbpVhQbVEGgWZIj1jwfIGhrF07BbKKgw==
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1NDU0MTExNjAsImV4cCI6MTU0NTQxNDc2MCwiYXVkIjoiY2hhcmdlc3BsIn0=.m6ZyoAWPmyumJZB6ZsXmKcpW1HgPZSsx/r7+8OHpXImXoy9A+Ww/RLkbpVhQbVEGgWZIj1jwfIGhrF07BbKKgw==
Waiting: 5542
Connecting to mqtt.googleapis.com
MQTT connecting ...
JWT now:
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1NDU0MTExNjcsImV4cCI6MTU0NTQxNDc2NywiYXVkIjoiY2hhcmdlc3BsIn0=.BOSlS4ZR4j/JatL7nA4NiQ3A6iTK0pF1R3XK2Tc0OBmsrfjLcsuLbUg2NM2hYWH0jbWBJAoVB7misybBnNKhvw==
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1NDU0MTExNjcsImV4cCI6MTU0NTQxNDc2NywiYXVkIjoiY2hhcmdlc3BsIn0=.BOSlS4ZR4j/JatL7nA4NiQ3A6iTK0pF1R3XK2Tc0OBmsrfjLcsuLbUg2NM2hYWH0jbWBJAoVB7misybBnNKhvw==
Waiting: 5083
Publish message
Connecting to mqtt.googleapis.com
MQTT connecting ...
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02

Waiting: 5309
Connecting to mqtt.googleapis.com
MQTT connecting ...
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02

Waiting: 5768
Publish message
Connecting to mqtt.googleapis.com
MQTT connecting ...
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02

Waiting: 5309
Connecting to mqtt.googleapis.com
MQTT connecting ...
projects/spl/locations/europe-west1/registries/pipger/devices/esp32_02

Waiting: 5268

Gior80, 2018-12-21:
I auto close the ticket but I leave here reason of problem: 

Inside pubSubClient.h. you need to modify:

// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 512. <------
#endif

// MQTT_KEEPALIVE : keepAlive interval in Seconds
#ifndef MQTT_KEEPALIVE
#define MQTT_KEEPALIVE 60 <-------
#endif

// MQTT_SOCKET_TIMEOUT: socket timeout interval in Seconds
#ifndef MQTT_SOCKET_TIMEOUT
#define MQTT_SOCKET_TIMEOUT 60 <--------
#endif

gguuss, 2018-12-21:
The advanced example is _intentionally_ [disconnecting and reconnecting to demonstrate reconnect](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-mqtt-advanced/Esp32-mqtt-advanced.ino#L139-L142). In practice, you would probably not want to do this, so I'll add a comment there.

gguuss, 2018-12-21:
For your second issue, I'll assume you're using PlatformIO... in this case, you will need to [specify a build rule to set the constants](https://github.com/googlecloudplatform/google-cloud-iot-arduino#user-content-known-issues).

closed 2018-12-23 by Gior80