GoogleCloudPlatform/google-cloud-iot-arduino #79: TRYING TO CONNECT SPARKFUN ESP-32 THING TO GOOGLE CLOUD IOT

jesuspisa, 2019-02-04:
I have some problems trying to connect the microcontroller sparkfun esp32 thing to google cloud iot. My objetive is to send telemetry from the groove data sensor to google cloud iot. In my case, I am using the DTH11 sensor to send both temperature and humidity.

Here is the error on Arduino Ide program as a result of executing the google cloud iot library skectch obviously changing some values to fulfill my requirements:

Connecting to WiFi
Waiting on time sync…
Connecting to cloudiotdevice.googleapis.com
Verifying certificate
Error: Certificate not verified! Perhaps the fingerprint is outdated.
GET /v1/projects/product12/locations/europe-west1/registries/iot-sample-registry/devices/iot-sample-device/config?local_version=0 HTTP/1.1

Connecting to cloudiotdevice.googleapis.com
Waiting for response…
headers received
7b

{
 “error”: {
 “code”: 401,
 “message”: “No \”authorization\” header found.”,
 “status”: “UNAUTHENTICATED”
 }
}

My doubt is when I create the certificates , Do I need to save them at any directory at my pc? It looks like it doesn’t find the certificates when is connecting to google cloud iot once I runned the script at Arduino IDE.

I can send you my scripts on Arduino IDE with those changes, so you can understand better my problem.

If it seems difficult to do, I could try to use Mongoose Os which is a firmware that supports ESP32 and can upload telemtry to google cloud iot, but I don't know if it supports groove sensors.

Thank you!

Gior80, 2019-02-05:
You need to create a payload with the information to be uploaded (json) and send the message to pubsub (look at send message part and payload in code) . At this point to do something with that data you need to create a cloud function (this is not part of the example)  triggered by pubsub writing . In this function you need to decode the json and do whatever u need (example write the data into a db and look at it by your webapp)

Gior80, 2019-02-05:
See here
https://cloud.google.com/functions/docs/tutorials/pubsub

jesuspisa, 2019-02-05:
Hi Gior 80, at which example of that tutorial are you refering to: https://github.com/GoogleCloudPlatform/nodejs-docs-samples/archive/master.zip

And another question if I use json, Do I need to use Arduino Ide program to connect to google cloud iot adding json. Or is it enough if I only use json to send telemetry to google cloud iot.

¡Thank you very much!

jesuspisa, 2019-02-05:
> Looks like you're using the ESP32 HTTP example, maybe try this example instead:
> 
> * [Google Cloud Esp32 IoT MQTT](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp32-lwmqtt)
> 
> We'll most likely be removing the non-lwmqtt examples soon to simplify the examples and make debugging customer issues easier.

------------------------------------------------------------------------------------------------

I run the example on Arduino IDE program, but I have errors compiling it. What could be the problem? Thank you!

esp32-mqtt.h:30:18: error: expected initializer before '*' token

 WiFiClientSecure *netClient;

                  ^

sketch\esp32-mqtt.h: In function 'void startMQTT()':

esp32-mqtt.h:83:51: error: 'netClient' was not declared in this scope

   mqttClient->begin("mqtt.googleapis.com", 8883, *netClient);

                                                   ^

sketch\esp32-mqtt.h: In function 'void setupCloudIoT()':

esp32-mqtt.h:127:3: error: 'netClient' was not declared in this scope

   netClient = new WiFiClientSecure();

   ^

gguuss, 2019-02-05:
@jesuspisa Have you installed the ESP32 SDK on Arduino?

gguuss, 2019-02-06:
Also, I recommend using the *lwmqtt exmaples, the ones with PubSub are difficult to debug.

jesuspisa, 2019-02-06:
@gguuss Yes, I have already installed the ESP32 SDK on Arduino but I don't know why appears that error executing the lwmqtt example. I executed another examples such as DHT11 sensor on Arduino and it works correctly.

jesuspisa, 2019-02-06:
@gguuss Is there another method to do the connection to google cloud iot? I read something about Node.js but I don't know if it supports Arduino.

Gior80, 2019-02-06:
And have u also installed the Library ?

jesuspisa, 2019-02-06:
@Gior80 Yes, I did it.  I executed the example of DHT11 sensor and it works properly. Does it mean that I installed the library?

And another question, in the previous answers when you referenced to json, Does you mean to create like a Node.js? Or update the skecth on Arduino program?

¡Thank you very much!

Gior80, 2019-02-06:
Lwmqtt

Gior80, 2019-02-06:
https://github.com/256dpi/arduino-mqtt

jesuspisa, 2019-02-06:
@Gior80 Is https://github.com/256dpi/arduino-mqtt the library should I install on Arduino?

¡Than you very much!

Gior80, 2019-02-06:
Yep this is

Gior80, 2019-02-06:
This is for lwmqtt version of the example 
For pubsub example you need the pubsub liebt library

jesuspisa, 2019-02-06:
@Gior80 Do you know what it refers the todo in those lines? 

String getJwt() {
  if (iss == 0 || time(nullptr) - iss > 3600) {  // TODO: exp in device
    iss = time(nullptr);
    Serial.println("Refreshing JWT");
    jwt = device->createJWT(iss);
  } else {
    Serial.println("Reusing still-valid JWT");
  }
  return jwt;
}

¡Than you very much!

Gior80, 2019-02-06:
This for renewing jwt key , don’care about

jesuspisa, 2019-02-06:
Hi @Gior80 ,
So in my case, If I want to read and send the temperature data, Should I do like that?

// To read the temperature
//iot-sample-pubsub is the topic once created at Google Cloud
//Temperature is float temperature = dht.readTemperature(); (I am using DHT11 sensor)
**void temperatureReceived**(String &iot-sample-pubsub, String &temperature) {
  Serial.println("incoming: " + iot-sample-pubsub + " - " + temperature);
}

//To send the data of temperature
**void startMQTT**() {
  mqttClient->begin("mqtt.googleapis.com", 8883, *netClient);
  mqttClient->onMessage(temperatureReceived);
}

¡Than you very much!

Gior80, 2019-02-06:
Onmessage void is triggered when a message is received from cloud  . Send message is used to send arguments (JSON ) using ex sendtelemetry method

jesuspisa, 2019-02-06:
Hi @Gior80 , so Do I need to copy the example of publishtelemetry in the example: https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp32-lwmqtt

void **publishTemperatureTelemetry**(Float temperature) {
  mqttClient->publish(device->getEventsTopic(), temperature);
}

Gior80, 2019-02-06:
Man .

Leave the example as it is . The only things you have to do is go here :

ing getDefaultSensor() {
  return  "Wifi: " + String(WiFi.RSSI()) + "db";
}

This method create a json strong  of Wi-Fi level signal : 
WiFi: 33 dB 

Replace string(WiFi.rssi) with string (functiontogetleveleofyoursensor) + “Celsius” 

That’s it .

Then I repeat . You can’t see anything on pubsub if you not create a function to read from it . 
Option  u can use the gcloud beta console to read the pubsub queue

jesuspisa, 2019-02-07:
¡Thank you @Gior80 for that better explanation!
I have executed the lwmqtt example but I have some errors once I compiled the sketch. I am having some problems with the library **WifiClientSecure.h**. Does anyone know @Gior80 , @gguuss how can I solve it? I am testing it on Windows 10.

C:\Users\Jesus\Documents\Arduino\libraries\WiFiClientSecure\src\ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, const char*, const char*, const char*, const char*, const char*)':

C:\Users\Jesus\Documents\Arduino\libraries\**WiFiClientSecure**\src\ssl_client.cpp:144:67: error: **'mbedtls_ssl_conf_psk' was not declared in this scope

                  (const unsigned char *)pskIdent, strlen(pskIdent));**

¡Thank you very much!

jesuspisa, 2019-02-12:
Hi @Gior80 , @gguuss I have already solved the problem with the WifiClientSecure library. I donwloaded the example of lwmqtt and executed it on Arduino Ide program. As a result of executing, this is what appears in the console of Arduino Ide.

[image]

I have made all steps to make the keys, but I don't know where I have to saved the .pem files. Do I have to save the private keys in the directory where the Arduino programs are saved? 

For example: 
C:\Users\Documents\Arduino\myEsp32examples

I don't know why I don't have a wifi connection once I compiled the example. I have wifi connection at home and I put all the wifi credentials correctly. 

Does anyone know how to solve this problem?

¡Thank you very much!

jesuspisa, 2019-02-13:
Hi guys, @Gior80 and @gguuss, I have already tested the Wifi connection on the Sparkfun ESP32 Thing and it works properly. I used the host "google.com" to test the connection.

The problem is that I don't know what domain introduce if I want to connect to google cloud. I was using the host : char * host="cloudiotdevice.googleapis.com";
But it is wrongly declared because it doesn't work. So, do you know what domain should I use to have a google cloud iot connection?

¡Thank you very much!

Gior80, 2019-02-14:
Checked of using right location ? In your registry settings?

jesuspisa, 2019-02-14:
Hi @Gior80 I'm from Spain and I have europe-west1 as region in my registry.

Gior80, 2019-02-14:
Yes , ok , i mean if u have set it up in code

jesuspisa, 2019-02-14:
@Gior80 Yes, I did it.

jesuspisa, 2019-02-14:
> You need to create a payload with the information to be uploaded (json) and send the message to pubsub (look at send message part and payload in code) . At this point to do something with that data you need to create a cloud function (this is not part of the example) triggered by pubsub writing . In this function you need to decode the json and do whatever u need (example write the data into a db and look at it by your webapp). Part u missing is that this example only publish the data on the pub/sub queue

Hi @Gior80 now the example lwmqtt is working correctly. I want to try to do what you said in one comment before but I don't know how to do it. Is there any tutorial?

¡Thank you very much!

jesuspisa, 2019-02-16:
Hi @Gior80 and @gguuss to send telemtry to pub/sub do you mean to use Node.js? 
Is it possible to do with Arduino?

¡Thank you very much!

Gior80, 2019-02-16:
Hello , this is something out of the example . I will suggest you to look google cloud function and pubsub trigger

jesuspisa, 2019-02-16:
Hi @Gior80, so Do I need to use the Arduino Ide Skecth (Esp32-lwmqtt) example or Do I need to use Node.js for sending telemtry to pub/sub without the arduino skecth?

¡Thank you!

Gior80, 2019-02-16:
No, example put the data on the pubsub , that’s it . You can take look in iot core console . If you want to do whatever with this data (ex put on a firebase instance or other ) u need to create a cloud function activated by pubsub trigger . This is something totally unrelated to example and to Arduino. There should be an article on medium about that

jesuspisa, 2019-02-18:
Hi guys, I need to store the data of DHT11 temperature/humidity sensor at Pub/sub. I modified a little bit the ESP32-lwmqtt skecth to publish the data on pub/sub. But, How I verify I'm doing things correctly? Do I need to see something at Pub/sub?

jesuspisa, 2019-02-19:
Hi @gguuss , Do you have deleted the pub/sub example on this repository?

¡Thank you!

gguuss, 2019-02-19:
Yes, it was confusing to have so many options and the lwMQTT client worked best in my tests.

gguuss, 2019-02-19:
As of v1.1.0b there are no longer HTTP examples or those using clients other than lwMQTT, Arduino retains an archive of the older libraries if you need them.

jesuspisa, 2019-02-20:
Hi @gguuss, Does ESP32-lwmqtt example write the data on pub/sub topic?

gguuss, 2019-02-20:
@jesuspisa Yes. The data will appear in the PubSub topic you have configured with your device registry.

jesuspisa, 2019-02-20:
Hi @gguuss , @Gior80 ,

I have made some changes to Esp32-lwmqtt skecth to obtain DHT11 temperature and humidity information:

**At esp32-mqtt.h:**
[image]

**At Esp32-lwmqtt, at loop() function**
[image]

My doubt is that  when I execute this command: 
**gcloud beta pubsub subscriptions pull --auto-ack iot-sample-sub**

I don't understand why in some cases only the data of the wifi is viewed , meanwhile in other cases both the wifi's and temperature's data . However, I haver never seen the humidity one.

[image]

How can I do to visualize wifi, temperature and humidity data?

¡Thank you very much!

gguuss, 2019-02-20:
>I don't understand why in some cases only the data of the wifi is viewed , meanwhile in other cases both the wifi's and temperature's data

I can think of a number of causes for this. I recommend you publish all of the sensor data simultaneously, e.g.:

    getSensorData() {
		return "{\n" + "\"wifi\":\"" + String(Wifi.RSSI()) + "db\"" + 
                "\n\"Temperature\":\"" + dht.readTemperature() + "C\"" + 
                "\n\"Hum\":\"" + dht.readHumidity() + "%\"\n}";
    }

> How can I do to visualize wifi, temperature and humidity data?

This is probably outside of the scope of this project to document here and it depends on how much data you'll really have.  For prototyping you could just write a client-side application that parses the JSON from PubSub and outputs a CSV. You could also write to a DB and then graph from the DB. You could plot the data in real-time from a PubSub reader.

jesuspisa, 2019-02-20:
Hi @gguuss , I have already modified the skecth, but I still have some problems.

**Modifications**

[image]

**Problems:**

[image]

I think that the first problem is because of the + sign at the getSensorData() function. The other problem was not detected before modifying the skecth, so I don´t know why have suddenly appeared.

¡Thank you very much!

I appreciate so much your help. I'm doing my best to solve those problems. :)

gguuss, 2019-02-21:
The code I shared was mostly as an example - I hadn't tested it. 

The first issue you're seeing is that the compiler cannot concatenate the types used in the string constants, e.g. `" \"Wifi\":'" "`. You can wrap the string initialization in a constructor, for example:

    return String("something") + String("Something else");

or:

    return String("something" + String("something else"));

Which may be helpful in resolving your issue with concatenation. There's a syntax error in the code I shared, `WiFi` in Arduino has a capitol F. That's why you're seeing the second problem.

jesuspisa, 2019-02-21:
Hi @gguuss , your proposals are working correctly!
Now, I'm trying to send the data from pub/sub into a database.
Do you know about any good tutorial?

¡Than you very much!

gguuss, 2019-02-26:
@jesuspisa I recommend you try the following:

* Use [Google Cloud Functions](https://cloud.google.com/functions/docs/) to publish PubSub data to a [Google Cloud SQL database](https://cloud.google.com/sql/docs/postgres/how-to).
* Chart the data out of your database using [any charting solution](https://medium.com/@bengarvey/making-charts-with-postgres-5700f4c65cc8)

It's really outside of the scope of this project to give individual instructions for this so I'm going to close this issue.

closed 2019-02-26 by gguuss