GoogleCloudPlatform/google-cloud-iot-arduino #83: ESP32 lwmqtt connection denied

kishoreinme, 2019-02-09:
Hello

I am trying out the v1.1.0 esp32-lwmqtt sample with this board with connection denied on mqtt_connect

https://www.cnx-software.com/2017/10/13/this-ttgo-board-combines-esp32-lora-radio-and-oled-display-for-just-10/
I  rechecked my device, registry, project and private key ciotc config. Infact I follow the same commands (even the names) in readme to tryout things first time by keeping  it simple. 

On the root certifcate part, he  openssl command in he instructions of ciotc returns a certifcate chain with two certificates and I am not sue which one to copy. I ust tried with the first certificate.

And also it says to copy the content to root.cert. I do not understand where the root.cert should be placed (or flashed into esp32 with some tool?!?!)

Any helpis appreciated. Thanks

jesuspisa, 2019-02-10:
Hi @kishoreinme I am trying to connect an sparkfun esp32 thing to google cloud iot too. I have followed the steps in this page: https://medium.com/@ludovic.emo/how-to-send-esp32-telemetry-to-google-cloud-iot-core-caf1a952020d

I saved all the certificates created by following the previous steps. I saved them in the directory where I saved my example Arduino Ide Progam. In my case C:\Documents\Arduino\my Esp32\Google Cloud Iot

Look at this picture: 

[image]

Then I uploaded at google cloud iot core the iot-sample-device-ec_public.pem to the device created at google cloud iot. 

[image]

Even if I did these steps, once I compiled my Arduino Ide Program I have this error: 

ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, const char*, const char*, const char*, const char*, const char*)':

C:\Users\Jesus\Documents\Arduino\libraries\WiFiClientSecure\src\ssl_client.cpp:144:67: error: 'mbedtls_ssl_conf_psk' was not declared in this scope

                  (const unsigned char *)pskIdent, strlen(pskIdent));

Do you have any compilation errors? Or Does it work perfectly to you? 
I compiled the Google Cloud Iot example appeared in the first link I shared: https://medium.com/@ludovic.emo/how-to-send-esp32-telemetry-to-google-cloud-iot-core-caf1a952020d

¡Thank you!

kishoreinme, 2019-02-11:
Hello @jesuspisa 
I did not face any compilation error with wifisecureclient. I just use the latest default libs.  And in my case I dont understand who will upload the keys to the board. 
Meanwhile ,  my use case  is fairly simple  with very few number of devices which are always powered, with enough bandwidth and did not need a full IOT suite with MQTT  like the GCP IOT core. I just wanted to try it out.
So I have moved to firebase with HTTP.

closed 2019-02-11 by kishoreinme