GoogleCloudPlatform/google-cloud-iot-arduino #84: Cant connect Adafruit ESP32 Huzzah

tyczj, 2019-02-12:
Going off the Esp32-lwmqtt sample I am trying to connect my Adafruit ESP32 Huzzah board. Previously I had it working with using a RSA256 key that I would get from an AppEngine endpoint but I want the device to be able to generate the JWT to remove the app engine endpoint call. 

I used openssl to create the public/private ec.pem keys 
Replaced the `private_key_str` with the `priv` value from running the command 

`openssl ec -in ec_private.pem -noout -text`

When I run the program I print the JWT to serial and put it in jwt.io and it says the JWT has a "Invalid Signature" so naturally the mqtt library kicks back saying back username or password.

Did I miss something?

Also what is the `root_cert` used for? I dont see it being used anywhere in the esp32 files

jesuspisa, 2019-02-12:
Hi @tyczj I am trying to compile the lwmqtt example correctly too. But, I have this problem when I executed it. 

[image]

Does you know how to solve it?

I don't have the error you have mentioned, so I don't know how to solve it.

Gior80, 2019-02-12:
Go lwmqtt guthun page , you can find in their error list readme

jesuspisa, 2019-02-12:
Hi @Gior80 ,
Do you refer to this link: 
https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/README.md ?

Gior80, 2019-02-12:
https://github.com/256dpi/lwmqtt/blob/master/include/lwmqtt.h#L243
https://github.com/256dpi/lwmqtt/blob/master/include/lwmqtt.h#L11

tyczj, 2019-02-12:
Back to my issue....

To rule out the known issue of

> Some private keys do not correctly encode to the Base64 format that required for the device bridge. If you've tried everything else, try regenerating your device credentials and registering your device again with

I went a created new public/private ec keys and I still have the issue

tyczj, 2019-02-14:
Fixed my issue, I missed configuring the time before creating the JWT

closed 2019-02-14 by tyczj