GoogleCloudPlatform/google-cloud-iot-arduino #86: Cannot connect MQTT using esp8266

ngoc233, 2019-02-19:
I can not connect MQTT and I receive an error is : " last error :-10  return code : 5". Could someone tell me where I am wrong?

gguuss, 2019-02-19:
This generally means that the client was disconnected due to authentication failure.

Did you set everything up in `ciotc_config.h`? You may want to try going through the [quickstart](https://github.com/googlecloudplatform/google-cloud-iot-arduino#quickstart) again.

ngoc233, 2019-02-21:
I'm setting up based on readme . I just change location. Like bellow
const char* project_id = "tactile-acrobat-231604";
const char* location = "asia-east1";
const char* registry_id = "atest-registry";
const char* device_id = "ngoc233";

const char* private_key_str =
    "c6:52:31:fc:2b:c6:88:4e:4a:29:ce:cd:1f:1b:c6:"
    "4a:47:36:0d:40:a4:ef:d2:30:95:11:61:2c:dc:a5:"
    "f6:58";

In the Serial, I have new error
 Success to open ca file
07:00:25.969 -> loaded
07:00:26.002 -> checking wifi...check wifi done
07:00:26.002 -> 
07:00:26.002 -> connecting...lasterror :-5
07:00:27.463 -> return code : 6

ngoc233, 2019-02-21:
**I see some problems in this code**
gcloud iot devices create atest-dev --region=us-central1 \
    --registry=atest-registry \
    --public-key path=ec_public.pem,type=es256

And I changed it like bellow

gcloud iot devices create ngoc233 \
  --project=project_id \
  --region=asia-east1 \
  --registry=atest-registry \
  --public-key path=ec_public.pem,type=es256

gguuss, 2019-02-26:
@ngoc233 If there's a bug in the README, let us know and we can update it. The region and other configuration for the device you create needs to be the same as the one you have configured in `ciotc_config.h`.

gguuss, 2019-02-26:
Make sure that `private_key_str` has 32 pairs in it. If not, you need to pad 00 or remove extra padding.

gguuss, 2019-03-12:
It looks like you're creating the device in the correct region and have a valid private key, I'm not sure why it is not working. Maybe try enabling a more verbose level of logging as described in [the updated README](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino). Closing as I cannot reproduce the issue.

closed 2019-03-12 by gguuss