GoogleCloudPlatform/google-cloud-iot-arduino #87: ESP32 Subscribe to Google Cloud IoT Topic

sambartek, 2019-02-22:
The current example sketch shows how to publish to pubsub, but does the current library have support for a device to subscribe to a separate topic in Google Cloud IoT?

closed 2019-02-26 by sambartek

l0c0luke, 2019-05-19:
is there an example for this?

l0c0luke, 2019-05-19:
I can find you subscribing here https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h

but if I try to subscribe to a PubSub topic, it keeps resetting the mqtt connection. 

I can receive a command to the device and see that reflected in the messageReceived... but not publishing a message to the topic. So I am trying to subscribe the device in the mqttConnect function.

gguuss, 2019-05-20:
## Publishing from IoT clients

You must [configure every PubSub topic 1:1 with a subfolder](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#publishing_telemetry_events).

## Subscribing IoT Clients

You can either subscribe to the configuration topic and receive configuration updates or you can subscribe to the commands topics and receive commands. For more complicated configurations, e.g. "publish anything from this topic as a command to a device" requires some work on your part like [mapping a Google Cloud Function between the two PubSub channels](https://cloud.google.com/community/tutorials/iot-device-to-device).

khawajamechatronics, 2019-06-11:
Can I get any example code to subscribe to the commands topics and receive commands

I am trying to understand the changes required without success