GoogleCloudPlatform/google-cloud-iot-arduino #88: keepalive/heartbeat on google iot core

alexodus, 2019-02-25:
In google cloud IoT Core the heartbeats (PINGREQ) are billed.
In my test (see the screenshot) with this librari I've 8 heartbeats/minute!!!

[image]

I tested with 60seconds (the default is 10 seconds) but the results are the same: 8 heartbeats/minute!!!
How to reduce this useless heartbeats?

alexodus, 2019-02-25:
this screen is about stackdriver on a single device:
[image]

gguuss, 2019-02-26:
Thanks for the report, confirming I'm seeing the issue and will see if I can figure out a way to disable.

gguuss, 2019-03-05:
The changes in this commit [explicitly increase the keepAlive / heartbeat](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/commit/44fd3068080e4f7300acc0dc4ae27d92b8d5c181) and will be in the next library release.

gguuss, 2019-03-09:
Which sample are you running?

gguuss, 2019-03-13:
Changes released in version 1.1.2, that said, I'm not seeing the heartbeat messages being generated by default.

closed 2019-03-13 by gguuss