GoogleCloudPlatform/google-cloud-iot-arduino #92: Help! The library does not work for me!

gguuss, 2019-03-13:
_Adding this issue here for tracking specific configuration issues from users._

Please include the following information:
* Device you're using (ex: ESP8266 WEMOS D1 MINI)
* Any relevant logs from ESP, see README.md for details on [setting logging levels](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/)
* Version of Arduino (ex: 1.8.1)
* Version of library / SDK (ex: ESP8266 By ESP Community 2.5.0)

MauriceWallbott, 2019-03-15:
Hello. thank you for the examples.
But I have some connection issues. 

Device you're using = ESP8266 WEMOS D1 MIN 
Any relevant logs from ESP, see README.md for details on setting logging levels
Version of Arduino = 1.8.8 
Version of library / SDK (ex: ESP8266 By ESP Community 2.5.0)
If you include what you're trying to do, it may also help to isolate the issue. This thread is intended for tracking resolutions so hopefully we'll get you sorted.

My Problem is:
- mqttClient->lastError() is 0
- mqttClient->returnCode() is 0

With this line: netClient->setInsecure();  in the mqttConnect method I have a valid connection.

Debug Output:
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687

SPIFFSImpl: allocating %zd+%z d+%z  d=%z   d bytes
SPIFFSImpl: mounting fs @300000, size=fb000, block=2000, page=100
SPIFFSImpl: mount rc=0
Success to open ca file
loaded
SPIFFS_close: fd=1

BSSL: Connection *will* fail, no authentication method is setup:wr 224 0

BizNinja, 2019-03-16:
Hello and thanks for the examples! I'm trying to get this to work on a ESP8266 sensor and am having issues seeing traffic flow through to my Google Cloud IoT registry. Followed all the steps laid out in the readme, as well as getting the SPIFFS loader operational. No error codes are returned, but my serial monitor reads these two lines repeating:
.0
0

Device: ESP8266  ESP-12E CP2102
Any relevant logs from ESP, see README.md for details on setting logging levels:
Ca cert success, and these next two lines repeating.
.0
0
(wifi connected return)
Version of Arduino: 1.8.8
Version of library / SDK: ESP8266 By ESP Community 2.5.0
Board Config:
[image]

Are there any reasons why this model might not play well with Google IoT? I'd also be interested if there are any models that have been tested successfully!

Gior80, 2019-03-16:
I don’t get why are u getting cheaper technology . Esp32 are so cheap , and working super good . Btw , can u post your source code ?

gguuss, 2019-03-21:
@MauriceWallbott Make sure that you uploaded the root certificate using the ESP8266 FS upload tool.

@BizNinja Good to see things worked smoothly on the ESP32, the sample code assumes there's FLASH on the board, which some of the barebones models don't include :)

smuehlst, 2019-05-13:
I'm trying to run the Esp8266-lwmqtt example with the following environment:

- Adafruit Huzzah ESP8266 breakout board
- Arduino 1.8.8 in combination with Visual Studio Code Arduino plugin
- ESP8266 SDK 2.5.1

I compiled the sketch and enabled the logging in WiFiClientSecureBearSSL.cpp. I have uploaded the root certificate to SPIFFS. When I run the sketch, I get the following logging output:

```
Connecting to WiFi
Waiting on time sync...
Success to open ca file
loaded

connecting...Refreshing JWT
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'

connected!

connecting...Refreshing JWT
```

This repeats forever. Note that the "connected!" message is misleading, as it appears although the connection actually failed. The following code does not check the return vaule of the call to `this->mqttClient->connect()`, which may fail with return value false, but does not cause `this->mqttClient->lastError()` to return a value different from `LWMQTT_SUCCESS`:

https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/9ba09d585607e50264502725a80ce4f42dd1c275/src/CloudIoTCoreMqtt.cpp#L157

```
void CloudIoTCoreMqtt::mqttConnect() {
...
    this->mqttClient->connect(device->getClientId().c_str(), "unused", getJwt().c_str(), false);

    if (this->mqttClient->lastError() != LWMQTT_SUCCESS){
```

See also `bool MQTTClient::connect()` in MQTTClient.cpp:

https://github.com/256dpi/arduino-mqtt/blob/2ba5912abf85842ea8bc0e0b7efa80feb8b8b669/src/MQTTClient.cpp#L260

```
  // connect to host
  if (!skip) {
    int ret = this->netClient->connect(this->hostname, (uint16_t)this->port);
    if (ret <= 0) {
      return false;
    }
  }
```

This returns without setting `this->_lastError`, which only happens further down in `MQTTClient::connect()`.

As a workaround I can add a  call `((WiFiClientSecure*)netClient)->setInsecure();` in `setupCloudIoT()`. Then the connection succeeds and the sketch publishes successfully.

smuehlst, 2019-05-16:
Update about my attempts to runthe Esp8266-lwmqtt example: After switching back to ESP8266 Arduino SDK version 2.4.2 the example works as expected without any further change.

gguuss, 2019-05-16:
@smuehlst Thanks for the update!

smuehlst, 2019-05-17:
@gguuss Are there known problems with the ESP8266 SDK  2.5.1, or should the Esp8266-lwmqtt example work in general with this version?

gguuss, 2019-05-17:
@smuehlst Good question, I'm seeing the same behavior as you're seeing with 2.5.1. There's a lot that was changed for the 2.5 releases including the switch to BEAR_SSL, unification of the file systems. There's probably some changes that are necessary to support this version.

smuehlst, 2019-05-17:
@gguuss Thanks, I'll stick to 2.4.2 then for the time being.

gguuss, 2019-05-17:
It may be worth trying to update the certificate bytes and retrying with 2.5.x but I won't be able to look further into it today.

MarcuCristian, 2019-05-23:
Hi, i m having a bit of a problem and i can't figure it out.
I'm using an esp8266 with 2.4.2 sdk version and a bme680 sensor. The sensor is doing readings just fine but when I run the example nothing gets send to the cloud and this loop always happens:

[image]

I also get this error from time to time:

[image]

 What am I doing wrong? Can you help me fix this please?

gguuss, 2019-05-23:
@MarcuCristian Are you calling the "sendTelemetry" function with the value you want to publish?

MarcuCristian, 2019-05-23:
@gguuss 
it lists 0 items. 
Also, using your answer on this post i tested the jwt token and figured it's not goodn the signature being invalid. 
[image]

I don't know how to fix this, i'll just start from scratch, think i did something wrong on the way!

gguuss, 2019-05-23:
@MarcuCristian It's very strange that you were seeing the configuration message get sent - you may want to double check the values in `ciotc_config.h` and maybe consider adding a new public / private key pair.

MarcuCristian, 2019-05-23:
@gguuss 
i think... i figured it out.
[image]

If this is it (and i think this was the problem) it's actually laughable.. sooo this is the first time I'm using the google cloud platform, and I have a trial account with the 300$ credit from the start... all good just that the account was not activated in order to use that credit so the functionality of the platform was off.
All i had to do was press an annotation on top of the screen just to make the account billable.
Thank you for your time solving my issue, all the best to you

gguuss, 2019-06-06:
@MarcuCristian Glad to hear it started working, I can follow up with the product team to see if there's a way we can get a 'grace' quota for our APIs.  Also will add a note to the README.

leandrotoledo, 2019-06-27:
Hi @gguuss, thank you for this library. I've been trying to set it up but I ran across some issues and I was hoping you could help me with some insight here.

Device you're using: **Adafruit Feather HUZZAH with ESP8266**
Version of Arduino: **1.8.9**
Version of library: **1.1.6**
Version of SDK: **ESP8266 By ESP8266 Community 2.4.2**
And billing is enabled for this project.

Similar to issue #99, I'm still having token expiration issues at v1.1.6:
```
23:09:19.001 -> Refreshing JWT
23:09:22.976 -> -10
23:09:22.976 -> LWMQTT_CONNECTION_DENIED
23:09:22.976 -> 5
23:09:22.976 -> LWMQTT_NOT_AUTHORIZED
23:09:22.976 -> Delaying 39230ms
23:10:02.230 -> Refreshing JWT
23:10:06.172 -> -10
23:10:06.172 -> LWMQTT_CONNECTION_DENIED
23:10:06.172 -> 5
23:10:06.172 -> LWMQTT_NOT_AUTHORIZED
23:10:06.172 -> Delaying 60000ms
```

I'm waiting to see if when the JWT expires the connection will correct itself; but it hasn't happened yet.
Let me know if I can help providing any further information.

extracheez, 2019-07-04:
> @MarcuCristian Are you calling the "sendTelemetry" function with the value you want to publish?

Hi Gus, where is the "sendTelemetry" function defined?

I was looking around for it but the link in one of your articles is broken:
"https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-ssd1306/Esp32-ssd1306.ino#L192-L231"

gguuss, 2019-07-04:
@extracheez It should [be linked here now](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h#L88).

extracheez, 2019-07-04:
> @extracheez It should [be linked here in the example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h#L88) and [here in the library](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/CloudIoTCoreMqtt.cpp#L53).

Ah publishTelemetry isn't working for me: 

So far I can connect the device to iot and send messages and config to the device. I can also see every time publishTelemetry is called that stackdriver logs it. Unfortunately when I try to pull I get "Listed 0 items".

[Here is a screen shot of my stackdriver logs, serial monitor and terminal](https://imgur.com/1rb38pP)

    Device you're using: ESP8266 D1 MINI (unsure of brand)
    Any relevant logs from ESP:
`08:24:22.932 -> connected!
08:24:23.297 -> incoming: /devices/atest-dev/config - test config
08:24:23.297 -> incoming: /devices/atest-dev/config - test config
08:25:09.606 -> Wifi: -73db
08:26:09.588 -> Wifi: -74db
08:27:01.558 -> incoming: /devices/atest-dev/commands - test command
08:27:09.618 -> Wifi: -73db
`
    Version of Arduino: 1.8.9
    Version of library: 2.4.2

Sorry if I'm missing something basic. I'm very new to all this.

gguuss, 2019-07-04:
@extracheez You need to both have a PubSub topic as well as a subscription to the PubSub topic. When you pull the subscription, you will see the messages appear.

extracheez, 2019-07-05:
> @extracheez It looks to me like you're REALLY close! You need to both have a PubSub topic as well as a subscription to the PubSub topic. When you pull the subscription, you will see the messages appear. As an alternative to using PubSub, you can publish state messages and they will appear in the Google Cloud Console. I'll update the link in my blog.

Success! Thank you!

Tarun1303, 2019-08-10:
> Hello and thanks for the examples! I'm trying to get this to work on a ESP8266 sensor and am having issues seeing traffic flow through to my Google Cloud IoT registry. Followed all the steps laid out in the readme, as well as getting the SPIFFS loader operational. No error codes are returned, but my serial monitor reads these two lines repeating:
> .0
> 0
> 
> Device: ESP8266 ESP-12E CP2102
> Any relevant logs from ESP, see README.md for details on setting logging levels:
> Ca cert success, and these next two lines repeating.
> .0
> 0
> (wifi connected return)
> Version of Arduino: 1.8.8
> Version of library / SDK: ESP8266 By ESP Community 2.5.0
> Board Config:
> [image]
> 
> Are there any reasons why this model might not play well with Google IoT? I'd also be interested if there are any models that have been tested successfully!
> 
> UPDATE:
> Thanks Gior80 - I was able to get my hands on an espressif ESP32, and the library is working great now. Going to roll forward with this and see if I can get a working POC. Thanks!

I am also facing the same issue can anyone help me out here

gguuss, 2019-08-16:
@Tarun1303 What version of the SDK and library are you using?  A few things worth checking:

* Is your EC key correctly registered in the IoT Console?
* Is your EC key valid / generated correctly?

You may want to try regenerating your key, registering it again, then double-checking all the parameters in `ciotc_config.h` - I have recently verified the library is working.

gguuss, 2019-11-12:
A few suggested resolutions in [this thread](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/123) for ESP8266:

* Disconnect before connecting

```
 if (!mqttClient->connected())
  {
    mqttClient->disconnect();
    ESP.wdtDisable();
    connect();
    ESP.wdtEnable(0);
  }
```

* Update to the latest ESP8266 community version 2.6.0

szundi, 2019-11-16:
> I don’t get why are u getting cheaper technology . Esp32 are so cheap , and working super good . Btw , can u post your source code ?

There are countries where esp32 vs esp-01s difference for example is significant compared to your wages. Or you may need like 100 devices for your company pet project. In this country. :) With esp-01s you can make whole packed installable solution of the price of an Esp32-wroom module itself.

Dewieinns, 2020-01-19:
I am trying to compile the bone stock example for the ESP8266 (sample Esp8266-lwmqtt) and I'm getting an error:

`esp8266_mqtt.h:45:1: error: 'X509List' in namespace 'BearSSL' does not name a type
`

I've been playing around adding libraries etc (a couple of 'BearSSL' libraries and their dependency) however haven't had any luck. I even removed any related library I could see and started the process over to be sure I had all dependencies without any luck. 

What am I missing folks?

Dewieinns, 2020-01-19:
> I am trying to compile the bone stock example for the ESP8266 (sample Esp8266-lwmqtt) and I'm getting an error:
> 
> `esp8266_mqtt.h:45:1: error: 'X509List' in namespace 'BearSSL' does not name a type `
> 
> I've been playing around adding libraries etc (a couple of 'BearSSL' libraries and their dependency) however haven't had any luck. I even removed any related library I could see and started the process over to be sure I had all dependencies without any luck.
> 
> What am I missing folks?
> 
> I'm sure it's something stupid but I've been scratching my head for a while now.

Of course I come across a solution moments after asking for help... For those who come across a similar issue the solution is: Update your board libraries. I remember now I had rolled back to a specific version for a tutorial I was trying to follow some time ago; I needed something newer.

balp, 2020-04-06:
Device you're using: Arduino UNO Wifi Rev 2
Any relevant logs from: 
Version of Arduino: 1.8.12
Version of library / SDK: WifiNINA-1.5.0, MQTT-2.4.7, Google Core IOT JWT- 1.1.9

I have problem compiling jwt, especially the encryption, nn. 

I'm missing cstring. I can't find any reference to cstring on the arduino pages.

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/andersar/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/andersar/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/andersar/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10812 -build-path /var/folders/qb/16t14spn3978fb_d4781nfsm0000gp/T/arduino_build_956202 -warnings=none -build-cache /var/folders/qb/16t14spn3978fb_d4781nfsm0000gp/T/arduino_cache_461935 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/andersar/Documents/Arduino/libraries/Google_Cloud_IoT_Core_JWT/examples/Esp8266-lwmqtt/Esp8266-lwmqtt.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/andersar/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/andersar/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/andersar/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10812 -build-path /var/folders/qb/16t14spn3978fb_d4781nfsm0000gp/T/arduino_build_956202 -warnings=none -build-cache /var/folders/qb/16t14spn3978fb_d4781nfsm0000gp/T/arduino_cache_461935 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/andersar/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/andersar/Documents/Arduino/libraries/Google_Cloud_IoT_Core_JWT/examples/Esp8266-lwmqtt/Esp8266-lwmqtt.ino
Using board 'uno2018' from platform in folder: /Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Using core 'arduino' from platform in folder: /Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Detecting libraries used...
/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -I/Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 /var/folders/qb/16t14spn3978fb_d4781nfsm0000gp/T/arduino_build_956202/sketch/Esp8266-lwmqtt.ino.cpp -o /dev/null
Alternatives for CloudIoTCore.h: [Google_Cloud_IoT_Core_JWT@1.1.9]
ResolveLibrary(CloudIoTCore.h)
  -> candidates: [Google_Cloud_IoT_Core_JWT@1.1.9]
/Users/andersar/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -I/Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/andersar/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/andersar/Library/Arduino15/packages/arduino/har [... cut ...]
                 from /Users/andersar/Documents/Arduino/libraries/Google_Cloud_IoT_Core_JWT/examples/Esp8266-lwmqtt/Esp8266-lwmqtt.ino:15:
/Users/andersar/Documents/Arduino/libraries/Google_Cloud_IoT_Core_JWT/src/crypto/nn.h:23:10: fatal error: cstring: No such file or directory
 #include <cstring>
          ^~~~~~~~~
compilation terminated.
Using library Google_Cloud_IoT_Core_JWT at version 1.1.9 in folder: /Users/andersar/Documents/Arduino/libraries/Google_Cloud_IoT_Core_JWT 
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
```

gguuss, 2020-04-06:
Yeah, this is not a supported board but you might be able to get it to work. My understanding is that the 328p does not have very much memory.

balp, 2020-04-07:
The UNO Wifi: It looks too weak, locks up calling, ecc_gen_pub_key. Same code (with WifiNINA) works on the MKR-WIFI1010.

gguuss, 2020-04-07:
@balp Interestingly enough it's using the ESP8266 for its modem, which is a supported target for this library. Unless you have a compelling reason to use this board, I'd recommend just using another one (e.g. MKR1000, ESP32/8266).

RyanGaudion, 2020-04-26:
Device you're using - ESP32
Version of Arduino - 1.8.12
Version of library / SDK - Current - Just pulled zip this morning

I'm trying to get the example sketch working. However, it get's stuck on "checking wifi". Please see serial monitor below:

```
Starting wifi
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Connecting to WiFi
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.69, MASK: 255.255.255.0, GW: 192.168.1.1
Waiting on time sync...
checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...checking wifi...

```
The only changes I have made is #154 as it wasn't building without this

netskink, 2021-08-08:
I am using latest as of 20210808 with MKR1000.  I get the following:

ClientId: projects/a-test-project/locations/us-central1/registries/MKR_GCP_Tutorial/devices/CESMKR1000
Waiting 60 seconds, retry will likely fail
Refreshing JWT
result is 0
mqttclient->lastError is -3
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.googleapis.com:8883
ClientId: projects/a-test-project/locations/us-central1/registries/MKR_GCP_Tutorial/devices/CESMKR1000
Waiting 60 seconds, retry will likely fail
Refreshing JWT
result is 0
mqttclient->lastError is -3
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.googleapis.com:8883
ClientId: projects/a-test-project/locations/us-central1/registries/MKR_GCP_Tutorial/devices/CESMKR1000
Waiting 60 seconds, retry will likely fail

repeated over and over

No changes to code other than ciotcl.h.  Would a wireshark capture help?