GoogleCloudPlatform/google-cloud-iot-arduino #96: 8266 refresh the connection every 70 minutes

ulysis01, 2019-03-18:
I can connect to GCP with 8266 using this library no problem. However approximately every 70 minutes (one hour and 10 minutes) I see that there is a new connection open.
I monitored the serial monitor and looks like every 70 minutes, it acts like wifi connection was lost and the device re-connects to the WiFi thus recalls the connect functions. The logs are pasted below.
This happens regardless of the RSSI and approximately every 70 minutes. To me there is a setting in the mqtt header that refreshes the JWT every 70 minutes. Can anyone advise?

checking wifi...
connecting...Refreshing JWT
connected!

gguuss, 2019-03-21:
Set the [expiration of the token when you create the JWT](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/CloudIoTCoreDevice.cpp#L48) and the password should expire within 20 minutes of the expiration time.

ulysis01, 2019-03-22:
Aha! thank you very much. 
I increased it to 2 hours and now I see a new connection request every 130 minutes. I would like to increase it even higher - perhaps only one refresh per day. Do you see a problem with that? Why did you want to set the refresh timer to 1h to begin with?

gguuss, 2019-03-22:
The reason you set the refresh time (expiration of the JWT) is to reduce the impact of a credential being compromised (e.g. CSRF on the web). The maximum allowable duration for a token to be valid is 24-hours and there are advantages to extending the duration that tokens are valid (e.g. less power used for recalculating JWT) that come with trade-offs (impact of abuse in case of credential leak). One hour is a pretty typical expiration duration when using JWTs but there isn't any specific reason I chose that duration.  Also of note, there is 20 minutes of "clock drift" that is allowed on the JWT, so your expiration won't be exactly the difference between the issue time and expiration.

closed 2019-03-22 by gguuss

ulysis01, 2019-03-22:
Very good. Thanks again for detailed response.