GoogleCloudPlatform/google-cloud-iot-arduino #99: Token expiring after 70 minutes.

thiagolc3, 2019-03-23:
Hi Gus.
6 months ago I was trying to use the mqtt examples for ESP32, but I stopped at the token expiration problem. Now I saw that you changed the examples a lot, it was easier to understand, this more didactic, but I still found the token expiration problem after 70 minutes. Now I'm using ESP8266, I can only refresh the token if I reset the device, where am I going wrong?

Thanks for attention.

gguuss, 2019-03-25:
The 70 minutes is probably coming from a combination of the 60 minute token expiration set in the JWT payload combined with the 20 minutes of clock drift allowed by the authentication backend.  If you're using the new examples, they should refresh the JWT token when it expires and subsequently re-authenticates the MQTT connection. Can you share any additional debugging information, as described in the [support issue](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/92)?

gguuss, 2019-03-25:
I'm able to repro the issue:

```

12:21:40.114 -> connecting...
12:21:40.777 -> connected!
incoming: /devices/arduino-taste/config - 1
12:21:40.843 -> incoming: /devices/arduino-taste/config - 1
incoming: /devices/arduino-taste/config - 1
incoming: /devices/arduino-taste/commands - test
incoming: /devices/arduino-taste/config - 1
incoming: /devices/arduino-taste/config - 0

12:56:16.156 -> connecting...
12:56:22.821 -> connected!
incoming: /devices/arduino-taste/config - 0
12:56:22.920 -> incoming: /devices/arduino-taste/config - 0
incoming: /devices/arduino-taste/commands - 123

13:25:43.999 -> connecting...Refreshing JWT
-10
13:25:54.012 -> LWMQTT_CONNECTION_DENIED
13:25:54.012 -> 5
13:25:54.012 -> LWMQTT_NOT_AUTHORIZED
13:25:54.012 -> Delaying 2120ms
-10
13:25:56.765 -> LWMQTT_CONNECTION_DENIED
13:25:56.765 -> 5
13:25:56.765 -> LWMQTT_NOT_AUTHORIZED
13:25:56.765 -> Delaying 4387ms
-10
13:26:01.772 -> LWMQTT_CONNECTION_DENIED
13:26:01.772 -> 5
13:26:01.772 -> LWMQTT_NOT_AUTHORIZED
13:26:01.772 -> Delaying 8864ms
-10
13:26:11.223 -> LWMQTT_CONNECTION_DENIED
13:26:11.223 -> 5
13:26:11.223 -> LWMQTT_NOT_AUTHORIZED
13:26:11.223 -> Delaying 18193ms
-10
13:26:30.055 -> LWMQTT_CONNECTION_DENIED
13:26:30.055 -> 5
13:26:30.055 -> LWMQTT_NOT_AUTHORIZED
13:26:30.055 -> Delaying 36681ms
-10
13:27:07.689 -> LWMQTT_CONNECTION_DENIED
13:27:07.689 -> 5
13:27:07.689 -> LWMQTT_NOT_AUTHORIZED
13:27:07.689 -> Delaying 60000ms
-10
13:28:08.296 -> LWMQTT_CONNECTION_DENIED
13:28:08.296 -> 5
13:28:08.296 -> LWMQTT_NOT_AUTHORIZED
13:28:08.296 -> Delaying 60000ms
-10
13:29:08.910 -> LWMQTT_CONNECTION_DENIED
13:29:08.910 -> 5
13:29:08.910 -> LWMQTT_NOT_AUTHORIZED
13:29:08.910 -> Delaying 60000ms
-10
13:30:09.491 -> LWMQTT_CONNECTION_DENIED
13:30:09.491 -> 5
13:30:09.491 -> LWMQTT_NOT_AUTHORIZED
13:30:09.491 -> Delaying 60000ms
-10
13:31:10.273 -> LWMQTT_CONNECTION_DENIED
13:31:10.273 -> 5
13:31:10.273 -> LWMQTT_NOT_AUTHORIZED
13:31:10.273 -> Delaying 60000ms
[... 258 lines cut ...]
14:23:43.522 -> LWMQTT_NOT_AUTHORIZED
14:23:43.522 -> Delaying 60000ms
-10
14:24:44.135 -> LWMQTT_CONNECTION_DENIED
14:24:44.135 -> 5
14:24:44.135 -> LWMQTT_NOT_AUTHORIZED
14:24:44.135 -> Delaying 60000ms
-10
14:25:45.078 -> LWMQTT_CONNECTION_DENIED
14:25:45.078 -> 5
14:25:45.078 -> LWMQTT_NOT_AUTHORIZED
14:25:45.078 -> Delaying 60000ms
Refreshing JWT

14:26:49.069 -> connected!
incoming: /devices/arduino-taste/config - 0
14:26:49.135 -> incoming: /devices/arduino-taste/config - 0

15:02:00.689 -> connecting...
15:02:08.249 -> connected!
incoming: /devices/arduino-taste/config - 0
15:02:08.316 -> incoming: /devices/arduino-taste/config - 0
```

However, you'll notice that the connection eventually corrects itself.  I'll see if I can get the connection to properly drop / resetart after the JWT expires.

Gior80, 2019-03-25:
Hi Man, 

had this problem for while and destroy my head on. If can help this seems direct connect in some way to MQTT protocol.  looks like need to fully reset the connection ( or go to a timeout.... ) to reconnect, in fact if you reset also wifi cons and reopen full wifi + ssl + iot core this work good ( a workaround could do to set a 24 Expiry and do that 1 time a day).  It would be SO useful to have acces to MQTT log ( I suppose ( I repeat I suppose....)  IOT core is a layer on standard IBM MQTT) to verifiy what's happening on server side.   Do you have ability to have a look  with you google internal super power?  ✌️

gguuss, 2019-03-26:
It looks to me that the current issue is that `getJwt` in each example is independently keeping state for the JWT issuance. I was able to successfully refresh the JWT and reconnect the client with the following change in ESP8266:

```c++
String getJwt() {
  if (iss == 0 || time(nullptr) - iss > 60) {  // TODO: exp in device
    // Disable software watchdog as these operations can take a while.
    ESP.wdtDisable();
    iss = time(nullptr);
    Serial.println("Refreshing JWT");
    jwt = device->createJWT(iss, 60); // TODO: fix force expiration
    ESP.wdtEnable(0);
  }
  return jwt;
}
```

There will be more refactoring necessary for making these less decoupled but I do believe that refresh is working, just out of sync with the token issue.

gguuss, 2019-03-26:
I found the bug -- the getJWT code was hardcoded to guard against refreshing the JWT too frequently.  

[This fix has been pushed to master](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/commit/b82ab9211c915443de0e6983fa453c441ace5817) and has been tested. This should release later this week.

I'll close the issue now, please reopen if you're still seeing the JWTs expire and not refresh after installing the latest library from the master branch.

closed 2019-03-26 by gguuss