GoogleCloudPlatform/google-cloud-iot-arduino #105: Error in esp32-mqtt.h

Bearg97, 2019-04-26:
Hey,

I am getting a compile error such as 'converting to 'String' from initializer list would use explicit constructor 'String::String(int, unsigned char)' in line 93 in esp32-mqtt.h file.

`void publishTelemetry(const char* data, int length) {
  mqtt->publishTelemetry(data, length);
}`

I tried a couple of things but I could not figure it out.

gguuss, 2019-04-29:
Thanks for the report!  I must have introduced an issue in the 1.1.5 release. You may need to include Arduino.h in some platforms.

gguuss, 2019-04-29:
What OS are you using and which version of the ESP32 SDK?

gguuss, 2019-05-17:
I'm unable to reproduce the issue, please feel free to add comments [in this issue](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/92) / try the suggestions from there.

closed 2019-05-17 by gguuss