GoogleCloudPlatform/google-cloud-iot-arduino #106: Prefer using the LTS certificate

gguuss, 2019-04-30:
As the project works now, we're using specific keys from the multiple root CAs provided by Google, we should instead find a smaller, canonical one.

gguuss, 2019-06-11:
Cert is available at mqtt.2030.ltsapis.goog

gguuss, 2019-06-11:
primary cert: https://pki.goog/gtsltsr/gtsltsr.crt
backup cert: https://pki.goog/gsr4/GSR4.crt

gguuss, 2019-06-20:
Update: MKR-1000 / ESP8266 do not like the provided cert.

gguuss, 2019-07-15:
Update: 1.1.7 adds support for the ESP8266 with LTS cert and domain.

koromkorom, 2019-09-29:
Hi, I have a quick question on this.

This happened recently and then of course ssl fails.:

"For added security, Google Trust Services may switch between the primary and backup certificates at any time without notice."

What would be the best way of handling this for ESP32 (and probably the others also)?

gguuss, 2019-11-05:
@koromkorom I'm guessing the best way of handling this would be to check both certificates.  This library is still pretty "experimental" and as far as I'm aware, we're just checking the primary certificate.

gguuss, 2019-11-22:
Closing issue as we're now using LTS cert and domain.  @koromkorom, I've opened another issue for adding support for the backup cert.

closed 2019-11-22 by gguuss