GoogleCloudPlatform/google-cloud-iot-arduino #108: Rationale for reference arguments in CloudIoTCoreMqtt constructor signature?

smuehlst, 2019-05-12:
The CloudIoTCoreMqtt constructor has three reference to pointer arguments:

```
CloudIoTCoreMqtt::CloudIoTCoreMqtt(
    MQTTClient *&_mqttClient, Client *&_netClient, CloudIoTCoreDevice *&_device){
  this->mqttClient = _mqttClient;
  this->netClient = _netClient;
  this->device = _device;
}
```
What's the rationale for declaring the arguments as references? This makes it impossible to pass non-lvalue pointers to the constructor.

What I would like to do in principle is to create a wrapper class like this:

```
class Wrapper {
private:
  CloudIoTCoreDevice device;
  CloudIoTCoreMqtt mqtt;
  MQTTClient mqttClient;
  WiFiClientSecure netClient;

public:
   Wrapper() : mqttClient(512),
    mqtt(&mqttClient, &netClient, &device) {}
}
```
But this is not possible because of the references.

gguuss, 2019-05-13:
It's been a while since I wrote the code encapsulating the MQTT clients. I am open to removing the reference arguments, I may have at one point been passing references to objects that were to be created by the client.

smuehlst, 2019-05-13:
I don't see any benefit in the reference arguments. With the wrapper class that I outlined above I want to avoid unnecessary dynamic allocations. Removing the reference from the argument should be compatible with all existing code.

gguuss, 2019-05-13:
@smuehlst ack. Makes sense, happy to make the change in the next update. If you'd be interested in reviewing, let me know.

Gior80, 2019-05-14:
I think I made the request so far , this was to have control on the client on sketch side

smuehlst, 2019-05-14:
@gguuss Thank you for accepting the proposal. I'll happy to review the change.

smuehlst, 2019-05-14:
@Gior80 Sorry, it is not clear to me what you mean. Are you in favor or against the proposal?

giorgio80, 2019-05-14:
hi @smuehlst I better explain. Actually in my sketch I initiate the connection outside the class as I need to verify connection status from main sketch (and manage reconnection), to use wifimanager to setup wifi and to use client for another library. I think this is useful . A solution btw could be to use both init method

gguuss, 2019-05-16:
[PR is here](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/109)

gguuss, 2019-05-17:
Fixed!

closed 2019-05-17 by gguuss