GoogleCloudPlatform/google-cloud-iot-arduino #117: jwt refresh

fabltd, 2019-07-11:
Hi

I have connectivity all working. 

Yet the IOT console shows the following error when the jwt token expires. -
 **mqtt: SERVER: The authorization token expired.** 

Should the token be refreshed before its expiry to prevent this error. 

The error does not impact operation.

gguuss, 2019-07-11:
Cool, just to clarify:

* Sample is working without issue
* Want to __avoid__ seeing token expiration error

IF that's the case, I am guessing we'd need to calculate a value for the token expiration that would be +- some portion of its expiration time and refresh it before the token expires as opposed to waiting for the error and refreshing on error?

fabltd, 2019-07-11:
Yes. Spot on all working. Just want to avoid the error. 

I think the resolution is correct. As I understand at the moment it waits for it to expire then regenerates on reconnect.

fabltd, 2019-07-14:
Are you planning a fix?

gguuss, 2019-07-14:
I'll happily accept the issue and will fix it when I can.

robin171, 2019-08-06:
Hi, 

I had the same issue. When the JWT expiration is set to 3600 seconds, Google disconnects after 1hour and 20 minutes with the JWT expired error. So just disconnect after 3600 seconds and reconnect again (jwt is refreshed upon connecting in the code). This solution is also prescribed by Google:

https://cloud.google.com/iot/docs/how-tos/credentials/jwts#refreshing_jwts

fabltd, 2019-11-05:
Hi Any update on this?

gguuss, 2019-11-05:
@fabltd The behavior you are seeing is happening because there is a +-20 minute "grace" period that can extend or reduce the expiration of the certificate. If you wanted to, on your own, explicitly refresh the JWT, just add some code to the main loop like:

```cpp
unsigned long lastMillis = 0;
unsigned long jwtMillis = millis();
void loop() {
  mqttClient->loop();
  delay(10);  // <- fixes some issues with WiFi stability

  if (!mqttClient->connected()) {
    connect();
  }

  // TODO: replace with your code
  // publish a message roughly every second.
  if (millis() - lastMillis > 60000) {
    lastMillis = millis();
    //publishTelemetry(mqttClient, "/sensors", getDefaultSensor());
    publishTelemetry(getDefaultSensor());
  }

  if (millis() - jwtMillis > (39*60*1000)) {
    getJwt();
  }
}

```

I'm not sure this should be the default behavior of the library.

robin171, 2019-11-06:
@gguuss In my experience you need to disconnect and reconnect in order to refresh the JWT

gguuss, 2019-11-12:
The JWT will refresh when it expires. There was a [bug in how TLS was connected that was fixed](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/commit/9725f918fb5526b1dd7d3e85911bf43ee0414833).

szundi, 2019-11-16:
@gguuss I have the same problem, just wanted to report this as a bug. I spend hours and figured out everything abow. Calling getJwt before the next keepalive should be enough according to the google article you mentioned about refreshing the JWT. But it isn't with 1.1.8, I just tried all above workaround attempts, only reconnect works... which is not nice. For that short period, there is no connection with the device and the service is down for some seconds, logs are cluttered with unmeaningful disonnects, etc.

I see this TLS bugfix with the added line with the skip variable, but honesty, I don't understand yet what implication it has with the situation. :) 

Can you please confirm that this TLS commit fixes this in the next release? 

My other issue is that I cannot really grasp the reasoning behind marking this as an enhancement as keeping a connection open is kind of pretty basic requirement in IoT. Frankly, without this workaround suggested above given user is struggling with server-side disconnected devices, that given the state of the 1.1.8 code, does not even recognise the disconnects and sits there for long time before some watchdog kicks in.

I know everyone should implement watchdogs, keepalives and such stuff, but aside from that this is a bug imho, with all my due respect to the team who made this code, that is awesome. You know, I just cannot really enter my door sometimes 😄 

Cheers!

gguuss, 2019-12-05:
I have just merged a new change into master that preemptively reconnects the client. Feel free to try it out, I'll probably stabilize it and release a new version soon.

gguuss, 2019-12-05:
Anecdotally, I have been running an ESP8266 since yesterday and am no longer seeing any `Authorization token expired` messages so this should be fixed in master and I'll close when I complete the next release.

szundi, 2019-12-05:
Sir, thank you, Sir. :)

DanielGarc, 2019-12-06:
> Anecdotally, I have been running an ESP8266 since yesterday and am no longer seeing any `Authorization token expired` messages so this should be fixed in master and I'll close when I complete the next release.
> 
> [image]

I started testing this afternoon, and I can confirm it's been running without the token expired log message.

gguuss, 2019-12-06:
I'll close the issue after releasing the library, this hopefully will happen early next week!

gguuss, 2019-12-11:
This should appear in the Arduino library as v1.1.9 when the release gets picked up, closing!

closed 2019-12-11 by gguuss