GoogleCloudPlatform/google-cloud-iot-arduino #121: Request example for MKR NB 1500

nicksav, 2019-08-07:
Guys
Looking to use with MKR NB 1500. Got it to the point:
[image]

But I am not sure If I need to install certificate, because there is no WIFI..
Please help!

gguuss, 2019-08-07:
It probably will not work out of the box as the network clients for the MKR NB 1500 seem to be different from the ones we have supported so far. You may want to take a look at [this SSL client example](https://www.arduino.cc/en/Tutorial/MKRNBExamplesNBSSLWebClient) and determine how to get / pass the NBSSLClient instead of the WiFiSSLClient in the initialization of the MQTT client in the board-specific `setupCloudIoT` function.  If you happen to get the board working, feel free to open a pull request with a new example / board.

gguuss, 2019-12-06:
Now that there's the NBSSLClient, this board may work with lwMQTT. As mentioned in Arduino's docs:

```
Porting a program from an Arduino Ethernet or WiFi101 library to an Arduino with the MKR NB 1500 should be fairly easy. 
```

gguuss, 2020-10-15:
Closing, this seems to be a duplicate of https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/142

closed 2020-10-15 by gguuss