GoogleCloudPlatform/google-cloud-iot-arduino #122: Exception running Esp8266_lwmqtt example

lvqthuong, 2019-08-13:
Today i'm running Esp8266_lwmqtt and got an exception after uploading the code  :

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40102b3a
EXCVADDR: 0x91413491

Decoding stack results
0x40203d36: NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 699
0x40203d3e: NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 699
0x40203e10: NN_Mult(unsigned int*, unsigned int*, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 179
0x40203d3e: NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 699
0x40203d3e: NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 699
0x40203de8: NN_Mult(unsigned int*, unsigned int*, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 175
0x40204650: omega_mul(unsigned int*, unsigned int*, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\secp256r1.cpp line 480
0x40203e8b: NN_ModMultOpt(unsigned int*, unsigned int*, unsigned int*, unsigned int*, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\nn.cpp line 456
0x402037f5: ecc_m_dbl_projective(point*, unsigned int*, unsigned char) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\ecc.cpp line 578
0x402038d8: ecc_win_mul(point*, unsigned int*, point*) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\ecc.cpp line 626
0x402039c8: ecc_gen_pub_key(unsigned int*, point*) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\crypto\ecc.cpp line 671
0x40204d81: CreateJwt(String, long long, unsigned int*, int) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\jwt.cpp line 131
0x4020bc28: free(void*) at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x4020b800: _umm_free(void*) at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x40208e84: String::copy(char const*, unsigned int) at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 211
0x40208ec4: String::String(char const*) at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 36
0x402026ae: CloudIoTCoreDevice::createJWT(long long, int) at C:\Users\nps_z\OneDrive\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\CloudIoTCoreDevice.cpp line 49
0x4020886d: Print::write(char const*) at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Print.h line 60
0x402088f8: Print::println() at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Print.cpp line 178
0x40208378: EspClass::wdtDisable() at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Esp.cpp line 103
0x40202245: getJwt() at D:\Build_Arduino\sketch/esp8266_mqtt.h line 63
0x40202e98: CloudIoTCoreMqtt::mqttConnect() at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/WString.h line 272
0x40202353: connect() at D:\Build_Arduino\sketch/esp8266_mqtt.h line 164
0x40202399: loop() at D:\google-iot-cloud\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 41
0x40202629: setup() at D:\google-iot-cloud\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 28
0x40209414: loop_wrapper() at C:\Users\nps_z\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 125
 ```

Here's some infomation which board i'm using  : 
+ NODEMCU v1.0 Lua - ESP8266 ESP12E
+ ESP8266 by Esp Community, version 2.5.1
[image]

jarrydharris, 2019-08-17:
Did you set up the SPIFFS upload utility?

zhaolewen, 2019-10-13:
I have the same exception and normally SPIFFS is uploaded

zhaolewen, 2019-10-15:
ok ... I finally decoded the exception and it's not exactly the same:
`Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40211c80: BearSSL::PrivateKey::isRSA() const at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\BearSSLHelpers.cpp line 728
EXCVADDR: 0x0059f896

Decoding stack results
0x40209eb4: esp_yield() at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4020a49f: delay(unsigned long) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x4020702a: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
0x40208709: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 231
0x40208c84: MQTTClient::connect(char const*, char const*, char const*, bool) at C:\Users\Zhao Lewen\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp line 262
0x402098fa: String::reserve(unsigned int) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 148
0x40209958: String::copy(char const*, unsigned int) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x40209aa8: String::operator=(String const&) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 259
0x40208df4: EspClass::wdtEnable(unsigned int) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Esp.cpp line 91
0x40209adc: String::String(String const&) at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 41
0x402022ee: getJwt() at C:\Users\ZHAOLE~1\AppData\Local\Temp\arduino_build_887470\sketch/esp8266_mqtt.h line 64
0x40203311: CloudIoTCoreMqtt::mqttConnect() at C:\Users\Zhao Lewen\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\CloudIoTCoreMqtt.cpp line 169
0x402083c1: BearSSL::WiFiClientSecure::available() at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 385
0x402076d2: BearSSL::WiFiClientSecure::connected() at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 259
0x402023fb: connect() at C:\Users\ZHAOLE~1\AppData\Local\Temp\arduino_build_887470\sketch/esp8266_mqtt.h line 162
0x40202456: loop() at C:\Users\Zhao Lewen\Egnyte\Private\zhao.lewen\Projects\IoT\gcp_iot_jwt/gcp_iot_jwt.ino line 42
0x40209ed5: esp_schedule() at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x40209f64: loop_wrapper() at C:\Users\Zhao Lewen\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125`

sphonala, 2019-10-24:
Update you Arduino core to latest or change your cores/esp8266/StackThunk.cpp
#define _stackSize (5600/4) to #define _stackSize (5750/4)
see this,
esp8266/Arduino#6143

gguuss, 2019-12-05:
I'll document this in the README, closing.

closed 2019-12-05 by gguuss