GoogleCloudPlatform/google-cloud-iot-arduino #123: connected but only after 5-6 attempts

comfortme, 2019-09-04:
Hello All
I managed to succesfully connect to iot-core, receive config and publish telemetry.
But only after 5-6 some times more failed attempts.
Device panics, restarts, tries again...finally connects if it's lucky.
Any idea why?

device - wemos mini esp12f
esp8266 version 2.5.2
latest versions of mqtt and iot libraries.

```
Waiting on time sync...
Success to open ca file
Success to open ca file

connecting...Refreshing JWT

Exception (3):
epc1=0x40219919 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4010209a depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb90 end: 3fffffc0 offset: 01a0
3ffffd30:  00000100 40219bba 401001c0 00002bae  
3ffffd40:  00000000 00000001 3ffe852c 00000001  
3ffffd50:  3fff2c7c 3fff092c 3fff092c 40211218  
3ffffd60:  00000000 4bc6a7f0 1126e978 00000001  
3ffffd70:  3fff2c7c 3fff092c 3fff092c 40208c1b  
3ffffd80:  0000012c 3fff422c 00000000 40208c5f  
3ffffd90:  00000000 00000000 3fff092c 00000001  
3ffffda0:  3ffffde0 3fff2c7c 3fff22dc 402110c8  
3ffffdb0:  00000137 00000000 3fff092c 4020800e  
3ffffdc0:  00000001 3fff22f0 00002e65 4020807a  
3ffffdd0:  00000000 3fff22f0 00000000 40201ed0  
3ffffde0:  00000000 00000005 401001c0 00002bae  
3ffffdf0:  00000000 00000000 3fff33e4 3ffffe90  
3ffffe00:  00000000 3fff2334 3fff22f0 4020103f  
3ffffe10:  00000137 00000000 3ffffe40 00000000  
3ffffe20:  00000137 00020f58 3fff22e0 4020804a  
3ffffe30:  003110c2 00000137 3fff22f0 40201f4c  
3ffffe40:  00000000 00000002 00000134 3fff22ac  
3ffffe50:  3fff3afc 00020f58 3fff22e0 3fff22ac  
3ffffe60:  3fff3afc 3fff2334 3fff22f0 4020128f  
3ffffe70:  40210054 3fff2e84 000100b4 40210006  
3ffffe80:  3ffe8ac4 402100ca 3fff3afc 00000000  
3ffffe90:  00000000 00000137 40211a80 ceb8e940  
3ffffea0:  3fff3afc 3ffe8ac4 3fff22f0 402082ef  
3ffffeb0:  40210054 3fff2e84 000100b4 40210006  
3ffffec0:  3ffe8ac4 402100ca 3fff3afc 00000000  
3ffffed0:  3fff2334 000003e8 3fffff30 4020908c  
3ffffee0:  40210054 3fff2e84 000100b4 40210006  
3ffffef0:  3ffe8ac4 402100ca 3fff3afc 402022be  
3fffff00:  3fff2e84 00000000 ff00002f 00000000  
3fffff10:  00000000 ff21002f 3fff35b4 3fff22ac  
3fffff20:  3ffef960 3fffff3c 3fff35b4 40202f19  
3fffff30:  3fff3afc 00ca00cf ffff22e8 3fff2e84  
3fffff40:  0054005f ff121363 3ffefaec 00000000  
3fffff50:  3fff2e84 3ffefaec 3ffe852c 3ffefaec  
3fffff60:  402487ce 042c1d80 3fff092c 3ffefa90  
3fffff70:  3fffdad0 00000000 3ffef940 402023b3  
3fffff80:  3fffdad0 00000000 3ffeef64 402023fd  
3fffff90:  3fffdad0 00000000 3ffefa60 40202695  
3fffffa0:  feefeffe 00000000 3ffefa60 4020951c  
3fffffb0:  feefeffe feefeffe 3ffe852c 401004d5  
<<<stack<<<
⸮QɆ-n⸮⸮0XConnecting to WiFi
Waiting on time sync...
Success to open ca file
Success to open ca file
[... 579 lines cut ...]

connecting...Refreshing JWT
-5
LWMQTT_NETWORK_FAILED_READ
0
OK
Delaying 2344ms
Refreshing JWT
-5
LWMQTT_NETWORK_FAILED_READ
0
OK
Delaying 5073ms
Refreshing JWT
-5
LWMQTT_NETWORK_FAILED_READ
0
OK
Delaying 10251ms
Refreshing JWT

connected!
incoming: /devices/esp/config - abc

```

DanielGarc, 2019-09-04:
I think I had a similar problem and adding a disconnect before connecting helped, let me know if that works for you.

``` c++
 if (!mqttClient->connected())
  {
    mqttClient->disconnect();
    ESP.wdtDisable();
    connect();
    ESP.wdtEnable(0);
  }
```

comfortme, 2019-09-08:
I've tried but got the same issue.

comfortme, 2019-09-09:
I decoded the stack trace if that helps;
```

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40210f78: BearSSL::PublicKey::isRSA() const at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src\BearSSLHelpers.cpp line 665
EXCVADDR: 0x006b6119

Decoding stack results
0x40207b98: BearSSL::WiFiClientSecure::_connectSSL(char const*) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 961
0x4020940c: esp_yield() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 91
0x402099eb: delay(unsigned long) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_wiring.cpp line 54
0x402068d9: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
0x40207d3d: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 231
0x40208250: MQTTClient::connect(char const*, char const*, char const*, bool) at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp line 262
0x40208e7e: String::reserve(unsigned int) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 148
0x40208edc: String::copy(char const*, unsigned int) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 211
0x4020902c: String::operator=(String const&) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 259
0x402083c0: EspClass::wdtEnable(unsigned int) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Esp.cpp line 91
0x40209060: String::String(String const&) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\WString.cpp line 41
0x402022be: getJwt() at C:\Users\Gonen\AppData\Local\Temp\arduino_build_17751\sketch/esp8266_mqtt.h line 70
0x40202f1d: CloudIoTCoreMqtt::mqttConnect() at C:\Users\Gonen\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src\CloudIoTCoreMqtt.cpp line 169
0x40206f6b: BearSSL::WiFiClientSecure::connected() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 259
0x402023b3: connect() at C:\Users\Gonen\AppData\Local\Temp\arduino_build_17751\sketch/esp8266_mqtt.h line 168
0x40202401: loop() at C:\Users\Gonen\Documents\Proje\arduino\brewer-could\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 42
0x40202699: setup() at C:\Users\Gonen\Documents\Proje\arduino\brewer-could\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 28
0x402094bc: loop_wrapper() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 125

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x402198b9
EXCVADDR: 0x4010209a

Decoding stack results
0x40219b5a: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x401001c0: millis() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_wiring.cpp line 186
0x402111b8: BearSSL::WiFiClientSecure::read() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 372
0x40208bbb: Stream::timedRead() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Stream.cpp line 34
0x40208bff: Stream::readBytes(char*, unsigned int) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\Stream.cpp line 211
0x40211068: Stream::readBytes(unsigned char*, unsigned int) at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/Stream.h line 92
0x40207fae: lwmqtt_arduino_network_read(void*, unsigned char*, unsigned int, unsigned int*, unsigned int) at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp line 42
0x4020801a: lwmqtt_arduino_timer_get(void*) at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp line 23
0x40201ed0: lwmqtt_read_from_network at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\lwmqtt\client.c line 83
0x401001c0: millis() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_wiring.cpp line 186
0x4020103f: lwmqtt_cycle_until at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\lwmqtt\client.c line 128
0x40207fea: lwmqtt_arduino_timer_set(void*, unsigned int) at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp line 11
0x40201f4c: lwmqtt_send_packet_in_buffer at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\lwmqtt\client.c line 186
0x4020128f: lwmqtt_connect at C:\Users\Gonen\Documents\Arduino\libraries\MQTT\src\lwmqtt\client.c line 387
[... cut ...]
0x402023b3: connect() at C:\Users\Gonen\AppData\Local\Temp\arduino_build_17751\sketch/esp8266_mqtt.h line 168
0x40202401: loop() at C:\Users\Gonen\Documents\Proje\arduino\brewer-could\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 42
0x40202699: setup() at C:\Users\Gonen\Documents\Proje\arduino\brewer-could\Esp8266-lwmqtt/Esp8266-lwmqtt.ino line 28
0x402094bc: loop_wrapper() at C:\Users\Gonen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266\core_esp8266_main.cpp line 125
```

sphonala, 2019-10-24:
Update you Arduino core to latest or change  your  cores/esp8266/StackThunk.cpp 
#define _stackSize (5600/4) to  #define _stackSize (5750/4)
see this,
https://github.com/esp8266/Arduino/issues/6143

gguuss, 2019-11-12:
Closing due to inactivity and adding note in support issue. @comfortme - let us know if the provided resolutions (update your SDK / fully disconnect) do not solve the issue.

closed 2019-11-12 by gguuss