GoogleCloudPlatform/google-cloud-iot-arduino #124: connection fails with mqtt->setUseLts(false);

stasysdr, 2019-09-14:
when mqtt->setUseLts(true); flag in row 182 in esp8266_mqtt.h is set to false (mqtt->setUseLts(**false**)), no connection can be established although "connect!" message is received at start.

gguuss, 2019-09-14:
Thanks for reporting this, I'll take a look. What port are you using (8883 or 443)? Also, I believe that the root certificate may need to be different when you change the server.

stasysdr, 2019-09-15:
tried both ports without success. used the same certificates as for tls though, which one would be correct?

stasysdr, 2019-09-27:
The problem was the certificate. Connection worked with "GlobalSign Root CA - R2".
For some non LTS method is faster by about 2 seconds, which is very important when battery saving is priority.

closed 2019-09-27 by stasysdr

gguuss, 2019-09-27:
@stasysdr Thank you for reporting this!