GoogleCloudPlatform/google-cloud-iot-arduino #125: Bug: CloudIoTCoreMqtt::mqttConnect ignores mqttClient's return code

matevarga, 2019-09-21:
https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/f6a33d1c5af4148cc403482d295a6de5ab855022/src/CloudIoTCoreMqtt.cpp#L169

The underlying connect method can often return `false` without setting `_lastError` properly. In this case mqttConnect logs 'connected!', while in reality connection has failed.

gguuss, 2019-11-12:
I've added a fix in the [log-fail-connect](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/log-fail-connect) branch.

gguuss, 2019-12-05:
Added detection in [this pr](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/139), the condition is generally caused by an SSL issue.

closed 2019-12-05 by gguuss