GoogleCloudPlatform/google-cloud-iot-arduino #127: ESP32 Example doesn't use provided rootCA

koromkorom, 2019-09-28:
Hi,

this issue is about the [ESP32 Example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/e230129e06fcee4b2c64349ca49ac0de63de13c7/examples/Esp32-lwmqtt)

The provided rootCA https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/e230129e06fcee4b2c64349ca49ac0de63de13c7/examples/Esp32-lwmqtt/ciotc_config.h#L58

is not used at all. 

Things were a little confusing, because I thought `setUseLts(true)` takes care of everything and the rootCA is somewhere hidden in the repo.

gguuss, 2019-11-12:
Just letting you know I'm aware of the issue. There are a few related issues here:

* Currently not using backup cert
* Not sure if the default is the LTS or the non-LTS cert.
* Still not sure we've properly documented how to generate these certificate bytes

meltdown03, 2019-12-09:
I can set the primary_ca OR the backup_ca cert by changing the `*netClient` to a `WifiClientSecure` pointer instead of a `Client` pointer and then using `netClient->setCACert(primary_ca);`

Unlike the ESP8266, I can't seem to create a list of certificates and can only use one or the other which is problematic.

Also noticed this:
https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/ef730971c398f41ee6bf0f4b829ba0baa84a5a01/src/CloudIoTCoreMqtt.cpp#L188-L190

It has the LTS server hardcoded and should change with what server is used.