GoogleCloudPlatform/google-cloud-iot-arduino #128: Issue in Publishing Data From ESP-8266 to Big Query

FreudVonDoom, 2019-10-03:
I am trying to Send Telemetry Data to GCP ( ESP -> IoT core -> Pub/Sub -> Data Flow -> Big Query). With the Schema of the table in Big Query being: 

_**Field name      | Type       | Mode     |** 
data                | STRING   | NULLABLE |  
telemetrydata | INTEGER | NULLABLE |_

I am getting the following error while Sending the data (using the "publishTelemetry(payload)" function) to Big Query: 
> Failed to serialize json to table row

What should the correct JSON format be of the variable `String payload` to send data to ( fields data and telemetrydata) big query ?

gguuss, 2019-10-03:
I have before made the mistake of missing quotes around keys:

**BAD**
```c++
String json = "{key:'value', key2:'value2'}";
```

**GOOD**
```c++
String json = "{'key':'value', 'key2':'value2'}";
```

May be worth trying before anything else.

FreudVonDoom, 2019-10-07:
It doesn't seem to work. It's giving me same error.

gguuss, 2019-10-07:
Nope, you would want to do that in your cloud function, dataflow job, or whatever script is migrating from PubSub to BigQuery.

gguuss, 2019-10-24:
@FreudVonDoom I've been demoing this from Python without issue. The following code migrates data from a DHT-22 sensor to BigQuery:

```python
def store_image_bq(project_id, sub_name, prefix, extension, duration, binary):
    """Store images in bigquery table"""
    from google.cloud import bigquery
    from datetime import datetime

    subscriber = pubsub.SubscriberClient()
    subscription_path = subscriber.subscription_path(project_id, sub_name)

    global count
    count = 0

    client = bigquery.Client()
    dataset_id = 'iotds'  # replace with your dataset ID
    table_id = 'events'  # replace with your table ID
    table_ref = client.dataset(dataset_id).table(table_id)
    table = client.get_table(table_ref)  # API request

    def callback(message):
        global count

        print('{}'.format(message))

        errors = []
        if message.attributes['deviceId'] == 'camera':
            image_data = message.data
            if binary:
                image_data = base64.b64encode(image_data)
                image_data = str(image_data, 'utf-8')

            print('Image: \n{}'.format(image_data))

            rows_to_insert = [
                (u'Photo', 0.0, datetime.now(), image_data),
            ]

            errors = client.insert_rows(table, rows_to_insert)  # API request

        elif message.attributes['deviceId'] == 'tempsensor':
            print('Temp data detected')
            data = message.data

            try:
                jsonData = json.loads(data)
                print('data: \n{}'.format(data))

                print('Found {}'.format(data))

                print(jsonData['T'])
                print(jsonData['H'])
                print(jsonData['I'])
                print(jsonData['D'])
                rows_to_insert = [
                    (u'tempsensor-TEMP', jsonData['T'], datetime.now(), b'', ""),
                    (u'tempsensor-HUM', jsonData['H'], datetime.now(), b'', ""),
                    (u'tempsensor-HEATINDEX', jsonData['I'], datetime.now(), b'', ""),
                    (u'tempsensor-DEWPOINT', jsonData['D'], datetime.now(), b'', ""),
                    (u'tempsonsor-COMFORT', 0.0, datetime.now(), b'', jsonData['Comfort'])
                ]
                print(rows_to_insert)

                errors = client.insert_rows(table, rows_to_insert)
            except json.decoder.JSONDecodeError:
                print('Dropping invalid message.')
                print(data)
                message.ack()
        else:
            strdata = message.data
            rows_to_insert = []

            rows_to_insert.push(
                (message.attributes['deviceId'], 0.0, datetime.now(), "",strdata))

            print('Data: \n{}'.format(image_data))

            errors = client.insert_rows(table, rows_to_insert)  # API request

        print(errors)
        message.ack()
        #query = '{}{}{}'.format(
        #    "insert into `intense-wavelet-343.iotds.events`",
        #    '(name, value, timestamp)',
        #    'values ("picture", 1.0, CURRENT_TIMESTAMP()')

    subscriber.subscribe(subscription_path, callback=callback)

    sleep_count = 0
    print('Listening for messages on {}'.format(subscription_path))
    while sleep_count < duration:
        time.sleep(1)
        sleep_count = sleep_count + 1
```

The following sketch shows how I'm generating the JSON to be transmitted over MQTT:

```cpp
      Serial.println("Publish temperature data...");
      String data = String(
        " {\"T\":" + String(newValues.temperature) + ", \"H\":" + String(newValues.humidity)
        + ", \"I\":" + String(heatIndex) + ", \"D\":" + String(dewPoint)
        + ", \"Comfort\":\"" + comfortStatus + "\"}");
        publishTelemetry(data);
    }
```

All of the values in the telemetry message are built around [this DHT-22 sample](https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/) installed from the Arduino library manager as `DHT sensor library for ESPx`.

Hope that helps.

closed 2020-10-15 by gguuss