GoogleCloudPlatform/google-cloud-iot-arduino #138: Error connecting

2color, 2019-11-24:
What is the correct `SSL Support` setting?

When configured to Basic I see the following in the console:
```
[hostByName] request IP for: mqtt.2030.ltsapis.goog
[hostByName] Host: mqtt.2030.ltsapis.goog IP: 64.233.184.206
:ur 1
:del
:ref 1
BSSL:_connectSSL: start connection
BSSL:Connection *will* fail, no authentication method is setup
:wr 115 0
:wrc 115 115 0
:ack 115
:rn 7
:rcl
:abort
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Unknown error code.'
```
With all SSL chipers are enabled I get the following error:

```
FATAL ERROR: BSSL stack overflow

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: bearssl
sp: 3fff1f14 end: 3fff1f94 offset: 0000
3fff1f14:  00002ad5 000000d5 00000004 deadbeef  
3fff1f24:  000000fe 00000000 00000000 00000000  
3fff1f34:  00000000 00000000 00000000 3fff1f64  
3fff1f44:  00000004 00000090 00000004 00002ad5  
3fff1f54:  00003a98 00000000 3fff0844 40209fc6  
3fff1f64:  3ffe8d4a 0000003e 00000000 40209fdc  
3fff1f74:  40258801 3fff35b0 deadbeef 40209055  
3fff1f84:  deadbeef 00000000 3fff0844 40204ce0  

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 0000
3ffffdd0:  00003a98 00000000 00000008 40207b9a  
3ffffde0:  000005f3 00000005 a7b12ee2 00000000  
3ffffdf0:  00000001 000016b7 3fff2dbc 40221e86  
3ffffe00:  3fff32bc 3fff20b4 3fff2dbc 00000001  
3ffffe10:  00000001 00000001 3fff0844 40207dd8  
3ffffe20:  3fff20b4 00000000 3fff0844 40207fbb  
3ffffe30:  00000023 3ffef974 00000001 40209a9f  
3ffffe40:  00000000 00000000 00000000 4020a0b7  
3ffffe50:  00000000 3fff2314 3fff0844 40206bb0  
3ffffe60:  000022b3 00000d50 3fffff00 3fff1fcc  
3ffffe70:  000022b3 3fff0844 3fff20b4 3fff1fcc  
3ffffe80:  000022b3 3fff0844 3fff20b4 402080d9  
3ffffe90:  4020c0b8 ceb8e940 4020c0b8 ceb8e940  
3ffffea0:  3fff2abc 3ffe8b4b 00000000 40208654  
3ffffeb0:  40208a4c 000000ce 3fffff30 4020940c  
3ffffec0:  3fff29dc 3ffeedac 3fffff30 402095cc  
3ffffed0:  00000000 3ffeedac 3fffff30 402095fc  
3ffffee0:  3ffeedac 3ffeedac 3ffef7d4 40208e44  
3ffffef0:  00000000 3ffeedac 3fffff30 40202222  
3fffff00:  3fff2384 00000000 80fe002f 6a6f7200  
3fffff10:  73746365 80fe002f 3fff206c 3fff1fcc  
3fffff20:  00000000 3ffef7d4 3fff206c 40202c99  
3fffff30:  3fff2abc 00ce00cf 00133400 3fff2384  
3fffff40:  0053005f 001334b8 3ffef974 00000000  
3fffff50:  3fff2384 3ffef974 0000000a 3ffef914  
3fffff60:  402417e2 042c1d80 3fff0844 3ffef914  
3fffff70:  3fffdad0 00000000 3ffef7b4 4020231a  
3fffff80:  3fffdad0 00000000 3ffeedd4 4020245c  
3fffff90:  3fffdad0 00000000 3ffef7d4 40202409  
3fffffa0:  feefeffe 00000000 3ffef8d4 40209bcc  
3fffffb0:  feefeffe feefeffe 3ffe8500 40100f39  
<<<stack<<<
```

[image]

gguuss, 2019-11-25:
I have had success with either configuration, what version of the SDK are you using?

gguuss, 2019-12-05:
I encountered the same issue and have gotten to the bottom of it:

* You need to enable all cyphers to connect
* You need to modify the memory allocated to BearSSL

To increase memory settings, see [this change to Arduino/cores/esp8266/StackThunk.cpp](https://github.com/esp8266/Arduino/issues/6811). I hope this gets improved in a future release.

closed 2019-12-05 by gguuss

2color, 2019-12-05:
Thanks @gguuss I'll give that a go.

How do I enable all cyphers?

DanielGarc, 2019-12-06:
> I encountered the same issue and have gotten to the bottom of it:
> 
> You need to enable all cyphers to connect
> You need to modify the memory allocated to BearSSL
> To increase memory settings, see this change to Arduino/cores/esp8266/StackThunk.cpp. I hope this gets improved in a future release.

I tested this today, as I was getting random resets until successfully connected. I discovered that the amount of bytes in the stack required to connect to the google lts server is more than the amount merged by [Expand BSSL stack #6819
](https://github.com/esp8266/Arduino/pull/6819)

Right now I'm getting a max usage of:
`BSSL stack: 6132`
and the commit increased the stack to:
`#define _stackSize (5900/4)`

> Thanks @gguuss I'll give that a go.
> 
> How do I enable all cyphers?

If you are using PlatformIO, cyphers are enabled by default
If you are using the Arduino IDE : 
[image]

2color, 2020-01-08:
Thanks @DanielGarc