GoogleCloudPlatform/google-cloud-iot-arduino #142: Compatibility with GSM/GPRS modules

jbarriosmuriel, 2019-12-12:
Hi

could you help me to configure these libraries with a gsm module, specifically the SIM800 connected to an esp32?

Thanks

gguuss, 2019-12-12:
Sure, happy to review any progress you make and give guidance on the required steps.

The best place to begin with understanding how Cloud IoT Core works is in the [documentation for concepts](https://cloud.google.com/iot/docs/concepts/).

If you from the device want to directly connect to Google Cloud IoT Core's MQTT bridge, you will need to figure out how to do SSL/TLS over GPRS to MQTT (8883).

If you want to connect to the HTTP bridge, you will need to get SSL/TLS over HTTP.

Regardless of the protocol you select, you will need to generate a JWT which is demonstrated in this library.

I have investigated the SSL/TLS aspect over GSM and got stuck. If you make any progress there, happy to hear from you.

An alternative approach to doing this would be to build a proxy that uses some type of identifier from your device to the remap messages from it to Cloud IoT Core. I haven't really looked into this but minimally, you could [use the existing examples](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge) to help with this mapping in C++/Python/NodeJS/Java.

jbarriosmuriel, 2019-12-13:
Thanks for your response, but my question is:  Can this library be adapted for use with the SIM800L module?

jorgemontalvao, 2019-12-16:
I have the same issue. I worked hard trying to customize it, but there are
too many dependencies hard-linked each other.
I prefer to work with Mongoose-OS
https://mongoose-os.com/docs/mongoose-os/cloud/google.md

During the "Quick Start" guide, instead of performing 7-configure-wifi
<https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md#7-configure-wifi>,
do the pppos <https://mongoose-os.com/docs/mongoose-os/api/net/pppos.md>
 configuration.

BUT I WOULD LOVE TO USE GOOGLE IOT instead of other non-Arduino solution

*Jorge Montalvão*

*+55(51)99145-4282*

*skype: montalvao84*
*hangout: jorge.montalvao@gmail.com <jorge.montalvao@gmail.com>*

gguuss, 2019-12-17:
> Thanks for your response, but my question is: Can this library be adapted for use with the SIM800L module?

Yes, this library can be adapted for use with GSM module. At a minimum the JWT function, used to generate the connection credential, would be reusable from this library as would the other helpers for generating your connection parameters. There is not an example for the SIM800 module because I ran into issues getting both MQTT support with TLS in a library for that module. If you can find such a library, then you should pretty easily be able to use this library for getting the connection parameters.

jbarriosmuriel, 2019-12-18:
Thanks you for the comments. Yes gguss, I will probe to modify this library to SIM800 or SIM808. If I get any results, I will share them with you. If someone gets it before letting us know please. 
Happy Holidays guys!

minusplusminus, 2020-01-18:
I think this should be possible with the [TinyGSM](https://github.com/vshymanskyy/TinyGSM) library. 

[TInyGSM modem](https://github.com/Xinyuan-LilyGO/TTGO-T-Call/blob/eacd2834b02766856b25c31c20b450f66b16a231/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino#L132)

This library uses this [MQTT library](https://github.com/256dpi/arduino-mqtt)

Because its base on the standard Arduino Client, It should just replacing the Wifi client with the TinyGSM client something like this: 
```

   Client* client = new TinyGsmClient(modem);
   setupCloudIoT(client);

void setupCloudIoT(Client * _tinyGSMClient) {
  device = new CloudIoTCoreDevice(
      project_id, location, registry_id, device_id,
      private_key_str);
       
  netClient =  _tinyGSMClient;
  mqttClient = new MQTTClient(512);
  mqttClient->setOptions(180, true, 1000); // keepAlive, cleanSession, timeout
  mqtt = new CloudIoTCoreMqtt(mqttClient, netClient, device);
  mqtt->setUseLts(true);
  mqtt->startMQTT();
}

```

https://github.com/256dpi/arduino-mqtt

xiskoct, 2020-01-20:
> I think this should be possible with the [TinyGSM](https://github.com/vshymanskyy/TinyGSM) library.
>
> [TInyGSM [...]

the return returned is: Settings incorrect or missing a cyper for SSL

xiskoct, 2020-01-20:
It seems that the AT Command certificate should be sent to the module? https://github.com/vshymanskyy/TinyGSM/blob/master/examples/more/SIM800_SslSetCert/SIM800_SslSetCert.ino

minusplusminus, 2020-01-21:
`Client* client = new GsmClientSecure(modem);` enables SSL support on sim800. I've tried, but yet no luck. Could not find any references of other people who got it working on the sim800

minusplusminus, 2020-01-22:
I think I know what the issue is:
[https://simcom.ee/documents/SIM800x/SIM800%20Series_SSL_Application%20Note_V1.02.pdf](https://simcom.ee/documents/SIM800x/SIM800%20Series_SSL_Application%20Note_V1.02.pdf)

> SIM800 series support SSL2.0, SSL3.0, TLS1.0

[https://cloud.google.com/iot/docs/how-tos/mqtt-bridge](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge)

> Initiate a TLS handshake over mqtt.2030.ltsapis.goog on port 8883 or 443. Use at least the following TLS features:
> 
> TLS 1.2
> P-256 with SHA-256 as the certificate key and hash algorithm
> TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256 using P-256 and uncompressed points for the cipher suite
> Server Name Indication
> DNS over TCP or UDP

SIM800 doesn't support TLS1.2

jyothi29, 2020-06-18:
Hello,
i want to send data from SIM 800A to server using pre shared keys(not certificate based).i have no idea how to configure the modem to establish TLS connection. someone please help me with this.

https://stackoverflow.com/questions/59543440/sending-data-with-pre-shared-keys-to-server-via-sim800a

felixback84, 2021-03-10:
Hello guys I have an ESP32 LilyGo T-Call SIM800 Series module with GSM and I want to send messages from it to GCP's IoT Core, for this I have used two TinyGSM libraries and
Google_Cloud_IoT_Core_JWT. Both do have been useful separately, that is, with TinyGSM I have managed to send messages via GSM to a server (using the examples it brings) and with Google_Cloud_IoT_Core_JWT I have managed to send MQTT messages via WiFi to GCP's IoT Core. However, the problem is when I try to do both at the same time. Since I always have the same error (the return returned is: Settings incorrect or missing a cyper for SSL) I have read this post and the one associated with it carefully and I understand that it is normal to use the Google_Cloud_IoT_Core_JWT library since it cannot establish a secure connection between the device and GCP, as I have seen and with I have investigated has something to do with the secure connection using an SSL certificate. Reading the comments of @Wiz-IO I have tried to intuit things, but nevertheless I have not been able to make it work, since although I know that he has succeeded, I cannot fully spin his suggestions on how to solve it, please do know What should I do? I ask you to help me, since it is important for a project of my company. For me everything happens in this part of the code. 

I take this code extract from the related post since I have tried to do almost exactly the same thing, with the same results

```
Client* client = new TinyGsmClient(modem);
setupCloudIoT(client);

void setupCloudIoT(Client * _tinyGSMClient) {
    device = new CloudIoTCoreDevice(
    project_id, location, registry_id, device_id,
    private_key_str);
         
    netClient =  _tinyGSMClient;
    mqttClient = new MQTTClient(512);
    mqttClient->setOptions(180, true, 1000); // keepAlive, cleanSession, timeout
    mqtt = new CloudIoTCoreMqtt(mqttClient, netClient, device);
    mqtt->setUseLts(true);
    mqtt->startMQTT();
}
```
------------------------------
**the return returned is: Settings incorrect or missing a cyper for SSL**

Thank you in advance in case any of those present in these post have managed to solve it

@Wiz-IO or somebaody I can have a more complete view of this piece of code or some further explanation, it is that especially in the final (begining in mqtt.secured()) part I do not know where the value of the variables comes from, many many tanks already:

https://github.com/Thalhammer/simcom_dam/issues/9#issuecomment-515372662

lptr, 2021-04-13:
@felixback84 I had a very similar issue recently with a simlar board. My understanding was that Google's MQTT servers require TLS 1.2, and the SIM800 on the LilyGo board can only do TLS 1.0. It works with WIFI because `WIFIClientSecure` supports TLS 1.2.

I managed to get things working by wrapping the `TinyGSMClient` in `SSLClient`, and the setup worked well: I could send and receive MQTT traffic over GPRS. However, the software implementation of TLS 1.2 was very slow, I couldn't actually call `mqtt->loop()` _and_ turn a stepper motor at the same time. (`mqtt->loop()` was calling `netClinet->connected()` which did a lot of SSL checks, causing everything to slow down a lot.) In the end I opted to drop the on-board GPRS, and install a separate 3G router and connect to that via WIFI.

Here's the removed code, maybe it helps you: https://github.com/kivancsikert/chicken-coop-door/pull/57