GoogleCloudPlatform/google-cloud-iot-arduino #144: ESP32 Setting incorrect or missing a cyber for ssl

chichunfung, 2020-01-14:
Settings incorrect or missing a cyper for SSL Error, is it private_key_str or root_cert wrong?

Error  message:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting wifi
Connecting to WiFi
Waiting on time sync...
checking wifi...
connecting...Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: 
Waiting 60 seconds, retry will likely fail
Refreshing JWT

gguuss, 2020-01-15:
I believe what you're encountering is a build setting. Check this section of the ESP8266 settings under the Arduino Tools menu:

[image]

The other potential issue is using the incorrect root certificate. You will need to [use the mqtt.2030.ltsapis.goog certificate described here](https://cloud.google.com/iot/docs/concepts/device-security#device_security_recommendations).

gguuss, 2020-02-06:
For ESP32, this could also be the root certificate not matching the server. If it's broken in the current release, please let me know. This could also indicate that there's an incorrect configuration (e.g. device ID, private key) - maybe try creating another device and connecting it again, sometimes that helps me debug in my own testing.

gguuss, 2020-03-24:
A final note, this is the error message that happens when your device is configured wrong.  I recommend making sure your device id is correct.

glush, 2020-04-15:
Seems it happens sometimes. But problem is not device configuration. 
Same run same app log:

---> IOT device id = <ID>
---> Success to primary open ca file
---> Success to open backup ca file

connecting...---> [getJwt] -> Refreshing JWT
---> [getJwt] -> Got new JWT - return
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/<PROJECT>/locations/us-central1/registries/<REGISTRY>-registry/devices/<ID>
Waiting 60 seconds, retry will likely fail
---> [getJwt] -> Refreshing JWT
---> [getJwt] -> Got new JWT - return
connected

Library connected!

olilogic77, 2020-04-20:
I get this error after about 60 minutes of successful running - I think its to do with the jwt refreshing after the initial period

mhanline, 2020-04-28:
I get this error intermittently as well. I'm seeing it after flakey wireless. I see in the Stackdriver logs:
message: "SERVER: The connection was closed because there is another active connection with the same device ID."
description: "ALREADY_EXISTS"
code: 6

If I wait a minute or two, it eventually connects again.

tyczj, 2020-05-06:
I get this issue too, sometimes it will reconnect just fine but other times it just keeps trying to reconnect infinitely. It usually takes a few days for it to fail like this though. It seems like something ends up being wrong with the JWT.

It will connect fine when initially powered up but letting it run for long periods I will eventually get this error

tyczj, 2020-05-06:
Based on the logging I have it looks like that the initial JWT is just being reused and not refreshed after it initially connects and an ssl error happens then it wont ever create another valid JWT

```
connecting...eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
connected

Library connected!
Reconnecting before JWT expiration

connecting...not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail

connecting...connected

Library connected!
Reconnecting before JWT expiration

[E][ssl_client.cpp:33] _handle_error(): [send_ssl_data():284]: (-80) UNKNOWN ERROR CODE (0050)
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail

connecting...not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail

connecting...not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail

connecting...not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail

connecting...not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODg3ODY1MDYsImV4cCI6MTU4ODc4NzQwNiwiYXVkIjoic21hcnQtaG9tZS0xNzQxMDEifQ==.qqpJr/8cXLQ2SRkLB4ufYJkc9gmF1UHsW0FbG81Nq+t2RLqWjKPomkIkdenKjaz0I7RHWlXzwTNdP6bBSTYK2Q==
Waiting 60 seconds, retry will likely fail
```

gguuss, 2020-05-07:
Thanks for the details on the issue, if I get some free cycles to put into this, I'll see if I can repro and come up with a fix. 

Something worth noting is that only one connection per device is supported at a time so if your device is somehow opening two connections, one of those connections gets dropped. As such, connecting two devices with the same ID / credentials creates unexpected behavior.

There's an ugly workaround for this and that is to use the HTTP API. I believe I removed support for HTTP from the library because MQTT is the preferred way (notification on configuration change, support for commands) but if you go back to 1.0.6 and earlier the code is still there and may be of use if you want to poll configuration on multiple devices.

raphael-bmec-co, 2020-05-27:
Hi Guys

I battled with this for ages on an ESP32 and long story short I was running out of stack. I would not be surprised if this is the cause for many of you.

meltdown03, 2020-05-30:
I'm having this issue on the new ESP32-S2 also. It works fine on my ESP8266 and ESP32. 

@raphael-bmec-co How did you increase the stack size on your ESP32?

raphael-bmec-co, 2020-05-30:
First check the available heap before and after the attempted connection using `ESP.getFreeHeap()`. From what I've seen it works if, before you make the connection attempt, you have 100K and it doesn if you have 50K. In between that I'm not sure.

Which version specifically of the ESP32-S2 do you have. You will need one that has PSRAM and then you will need to enable it in the board config.

meltdown03, 2020-05-30:
I have the WROVER ESP32-S2 and PSRAM is enabled. It's got 92020 bytes
FreeHeap right before mqttClient->connect() function on line 52 of
CloudIoTCoreMqtt.cpp. FreePsram is over 2 million bytes free.

The specific error is:
 -9
LWMQTT_MISSING_OR_WRONG_PACKET
6
LWMQTT_UNKNOWN_RETURN_CODE

raphael-bmec-co, 2020-05-30:
Based on the FreeHeap, that probably isn't the issue - the error also doesn't look familiar. Can you share the logs from google cloud platform side?

meltdown03, 2020-05-30:
```
{
 insertId: "1p1wa8pfl1vn9q"  
 jsonPayload: {
  disconnectType: "SERVER"   
  eventType: "DISCONNECT"   
  protocol: "MQTT"   
  resourceName: "projects/espressif-sensors/locations/us-central1/registries/espressif-devices/devices/2809118778375477"   
  serviceName: "cloudiot.googleapis.com"   
  status: {
   code: 9    
   description: "FAILED_PRECONDITION"    
   message: "The connection broke or was closed by the client."    
  }
 }
 labels: {
  device_id: "esp32-s2"   
 }
 logName: "projects/espressif-sensors/logs/cloudiot.googleapis.com%2Fdevice_activity"  
 receiveTimestamp: "2020-05-30T11:36:51.480828232Z"  
 resource: {
  labels: {
   device_num_id: "2809118778375477"    
   device_registry_id: "espressif-devices"    
   location: "us-central1"    
   project_id: "espressif-sensors"    
  }
  type: "cloudiot_device"   
 }
 severity: "ERROR"  
 timestamp: "2020-05-30T11:36:50.451013641Z"
}

meltdown03, 2020-05-30:
All the errors are the same. I just switched the loggind to Debug and here's the CONNECT log entry:
```
{
  "insertId": "gxcikafj11vv6",
  "jsonPayload": {
    "resourceName": "projects/espressif-sensors/locations/us-central1/registries/espressif-devices/devices/2809118778375477",
    "eventType": "CONNECT",
    "status": {
      "code": 0,
      "description": "OK"
    },
    "protocol": "MQTT",
    "serviceName": "cloudiot.googleapis.com"
  },
  "resource": {
    "type": "cloudiot_device",
    "labels": {
      "project_id": "espressif-sensors",
      "device_registry_id": "espressif-devices",
      "location": "us-central1",
      "device_num_id": "2809118778375477"
    }
  },
  "timestamp": "2020-05-30T11:43:07.511945996Z",
  "severity": "INFO",
  "labels": {
    "device_id": "esp32-s2"
  },
  "logName": "projects/espressif-sensors/logs/cloudiot.googleapis.com%2Fdevice_activity",
  "receiveTimestamp": "2020-05-30T11:43:08.553950405Z"
}
```
and the FAILED:
```
{
  "insertId": "gxcikafj11vv7",
  "jsonPayload": {
    "disconnectType": "SERVER",
    "resourceName": "projects/espressif-sensors/locations/us-central1/registries/espressif-devices/devices/2809118778375477",
    "eventType": "DISCONNECT",
    "serviceName": "cloudiot.googleapis.com",
    "protocol": "MQTT",
    "status": {
      "message": "The connection broke or was closed by the client.",
      "description": "FAILED_PRECONDITION",
      "code": 9
    }
  },
  "resource": {
    "type": "cloudiot_device",
    "labels": {
      "device_num_id": "2809118778375477",
      "project_id": "espressif-sensors",
      "device_registry_id": "espressif-devices",
      "location": "us-central1"
    }
  },
  "timestamp": "2020-05-30T11:43:08.469722790Z",
  "severity": "ERROR",
  "labels": {
    "device_id": "esp32-s2"
  },
  "logName": "projects/espressif-sensors/logs/cloudiot.googleapis.com%2Fdevice_activity",
  "receiveTimestamp": "2020-05-30T11:43:08.553950405Z"
}

raphael-bmec-co, 2020-05-30:
Right. Looks like the issue is on the device side. 

`LWMQTT_MISSING_OR_WRONG_PACKET` seems to come from `client.c's` `lwmqtt_connect()`. You'll have to try printing some logs from there to see what is happening. Beyond that, I am afraid I cannot be much help. Let us know if you get it.

meltdown03, 2020-05-30:
I started debugging `client.c` functions, in the `lwmqtt_cycle_until(client, &packet_type, 0, LWMQTT_CONNACK_PACKET);` function (line 387), the `packet_type` stays `LWMQTT_NO_PACKET` the entire time. I was trying to use a BME280 via I2C to send temp, humidity, and pressure data.
I finally got it to work by sending phony data and not reading the BME280 at all.
There must be a resource issue with WiFi and I2C together. I'll try the SPI bus next.
At least I know it's not this library or the MQTT library.

glush, 2020-05-30:
I think it's not I2C issure. I'd prepare empty sketch just sending dummy data to Cloud IOT Core. Also I put some debug print inside CloudIoTCoreMQTT.cpp. Problem appears randomly just after JWT refresh. After long attempts to connect with different error codes from lv_mqtt it's usually connected back and works till next jwt refresh. 

---> Send event [1590849439]
---> Send event [1590849449]
---> Send event [1590849460]
---> Send event [1590849470]
Reconnecting before JWT expiration
Refreshing JWT
---> new JWT = eyJhbG...ko+PA==

connecting...Refreshing JWT
---> new JWT = eyJhbG...0rvJw==
not connected
SYNC - Settings incorrect or missing a cyper for SSL
-6
LWMQTT_NETWORK_FAILED_WRITE
6
LWMQTT_UNKNOWN_RETURN_CODE
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST
Waiting 60 seconds, retry will likely fail
Refreshing JWT
---> new JWT = eyJhbG...RedPg==
---> !SKIP HOST = mqtt.2030.ltsapis.goog PORT = 8883
connected

Library connected!
---> Send event [1590849548]
incoming: /devices/ITERMOTEST/config - 
Wifi = OK MQTT = BAD

connecting...Refreshing JWT
---> new JWT = eyJhbG...uNyiA==
---> !SKIP HOST = mqtt.2030.ltsapis.goog PORT = 8883
not connected
-9
LWMQTT_MISSING_OR_WRONG_PACKET
6
LWMQTT_UNKNOWN_RETURN_CODE
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST
ASYNC - No internet or Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST

aborting mqtt connection attempt, lets rety later...	Library not connected!
---> Send event [1590849559]
Wifi = OK MQTT = BAD

connecting...Refreshing JWT
---> new JWT = eyJhbG...4W2iw==
---> !SKIP HOST = mqtt.2030.ltsapis.goog PORT = 8883
not connected
-9
LWMQTT_MISSING_OR_WRONG_PACKET
6
LWMQTT_UNKNOWN_RETURN_CODE
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST
ASYNC - No internet or Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST

aborting mqtt connection attempt, lets rety later...	Library not connected!
Wifi = OK MQTT = BAD

connecting...Refreshing JWT
---> new JWT = eyJhbG...XPJow==
[... 231 lines cut ...]
LWMQTT_CONNECTION_DENIED
5
LWMQTT_NOT_AUTHORIZED
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST
ASYNC - No internet or Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/ithermo/locations/us-central1/registries/vilia-registry/devices/ITERMOTEST

aborting mqtt connection attempt, lets rety later...	Library not connected!
Wifi = OK MQTT = BAD

connecting...Refreshing JWT
---> new JWT = eyJhbG...JGMsw==
---> !SKIP HOST = mqtt.2030.ltsapis.goog PORT = 8883
connected

Library connected!
---> Send event [1590849651]
incoming: /devices/ITERMOTEST/config - 
incoming: /devices/ITERMOTEST/config - 
---> Send event [1590849662]
---> Send event [1590849672]
---> Send event [1590849682]
---> Send event [1590849693]

meltdown03, 2020-05-30:
I'm doing that now with the dummy data, but I didn't have the same issue as you @glush. I'm also using the ESP32-S2

meltdown03, 2020-05-30:
I switched to a different BME280 library called JWA BME280, which is basically a wrapper for Bosch's code for the sensor. It defaults at 100k i2c freq. I had to correct a couple typedefs and change the i2c pins, but it works now for me. It's a much lighter library. So maybe that helps plus the lower I2C frequency.

AkeJohansson, 2020-05-30:
Hi, I'm having exacty the same problem but I use MKR1000.  
It's usually connected back and works till next jwt refresh.
This is my output from serial monitor.

Don't understand why the code connect to "mqtt.2030.ltsapis.goog:8883" I set mqtt->setUseLts(false); can't connect because it's not possible to install the certifacate to that domain.
I can only use mqtt.googleapis.com:443 

5:34:32.645 -> Library connected!
15:34:32.738 -> incoming: /devices/test1/config - 
15:34:32.831 -> mqttClient connected!!
15:39:24.728 -> Reconnecting before JWT expiration
15:39:24.728 -> Refreshing JWT
15:39:32.056 -> 
15:39:32.056 -> connecting...Refreshing JWT
15:39:39.339 -> not connected
15:39:39.339 -> Settings incorrect or missing a cyper for SSL
15:39:39.339 -> Connect with mqtt.2030.ltsapis.goog:8883
15:39:39.339 -> ClientId: projects/willo-project/locations/europe-west1/registries/willo/devices/test1
15:39:39.339 -> Waiting 60 seconds, retry will likely fail
15:40:39.297 -> Refreshing JWT
15:40:47.103 -> connected
15:40:47.103 -> 
15:40:47.103 -> Library connected!
15:40:47.197 -> incoming: /devices/test1/config - 
15:40:47.244 -> mqttClient connected!!
15:45:39.235 -> Reconnecting before JWT expiration
15:45:39.235 -> Refreshing JWT
15:45:46.511 -> 
15:45:46.511 -> connecting...Refreshing JWT
15:45:53.799 -> not connected
15:45:53.799 -> Settings incorrect or missing a cyper for SSL
15:45:53.799 -> Connect with mqtt.2030.ltsapis.goog:8883
15:45:53.799 -> ClientId: projects/willo-project/locations/europe-west1/registries/willo/devices/test1
15:45:53.799 -> Waiting 60 seconds, retry will likely fail
15:46:53.796 -> Refreshing JWT
15:47:01.635 -> connected
15:47:01.635 -> 
15:47:01.635 -> Library connected! 
15:47:01.728 -> incoming: /devices/test1/config - 
15:47:01.823 -> mqttClient connected!!
15:51:53.689 -> Reconnecting before JWT expiration
15:51:53.689 -> Refreshing JWT
15:52:00.995 -> 
15:52:00.995 -> connecting...Refreshing JWT
15:52:08.273 -> not connected
15:52:08.273 -> Settings incorrect or missing a cyper for SSL
15:52:08.273 -> Connect with mqtt.2030.ltsapis.goog:8883
15:52:08.273 -> ClientId: projects/willo-project/locations/europe-west1/registries/willo/devices/test1
15:52:08.273 -> Waiting 60 seconds, retry will likely fail
15:53:08.236 -> Refreshing JWT
15:53:16.030 -> not connected
15:53:16.030 -> Settings incorrect or missing a cyper for SSL
15:53:16.030 -> Connect with mqtt.2030.ltsapis.goog:8883
15:53:16.030 -> ClientId: projects/willo-project/locations/europe-west1/registries/willo/devices/test1
15:53:16.030 -> Waiting 60 seconds, retry will likely fail
15:54:16.014 -> Refreshing JWT
15:54:23.761 -> connected

meltdown03, 2020-05-30:
It probably is connecting to the regular mqtt server. The code shows the
LTS address no matter what. It's hardcoded into the logConfiguration()
function which is incorrect.

raphael-bmec-co, 2020-05-30:
There is a clear pattern here that it fails to connect the first time after the JWT update but then succeed. 

I'm not in front of the code right now, but in the esp 32 example there is a 10ms delay between calling loop and calling connect. 

I have observed similar behaviour to that which you have reported when removing this delay.

So I suspect there needs to be a delay (or a longer delay) between the JWT refresh and the reconnect.

Maybe you can take a shot at modifying the library to add a delay and report back?

Kind regards.

AkeJohansson, 2020-05-30:
Ok I try to modify the library. 
I'll be back :-)

glush, 2020-05-30:
I'd try to put 5 sec delay after JWT refresh. It does not help.

AkeJohansson, 2020-05-30:
OK.  :-(

raphael-bmec-co, 2020-05-30:
It seems as though you have set the JWT timeout to 5 seconds. There are limits here (https://cloud.google.com/iot/quotas) that suggest you need to wait 60s. Please try setting the timeout to 90s and let's see what happens.

meltdown03, 2020-05-30:
Try changing the keepAlive timeout (Esp**-mqtt.h - mqttClient->setOptions) to
slightly more than your PUBLISH_INTERVAL. Mine are 200 and 18000 (180 secs)
respectively.
Set the delay in main.cpp after mqtt->loop() to 50ms.

In CloudIoTCoreMqtt::loop(), change mqttConnect(true) to
mqttConnect(false). <<<THIS REALLY HELPED

I'm getting a new JWT key every 6 mins and so far no errors! It's gone
through 5 times only, but it would always error out before.

AkeJohansson, 2020-05-30:
Ok I'll give it a try.

Med vänliga hälsningar

Åke Johansson
ake.h.johansson@bravida.se
Tekniker / Technician

Bravida Sverige AB

[http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Logga.jpg]<http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Logga.jpg>
Telefon +46730-477647

Vx. +46470-377 00

Hjalmar Petris väg 32, Växjö

Besök oss gärna på www.bravida.se<https://www.bravida.se/>
Visit us at www.bravida.se/en/<https://www.bravida.se/en/>
Facebook.com/bravida.se<http://www.facebook.com/bravida.se>

[http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Bravida_mailbanner.jpg]<http://brain.corp.bravida.com/Global/Sverige/Aktuellt/Bilder/2016/Bravida_mailbanner%20%20liten%20.jpg>
 [http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/SBA_Badges_2018_mailsignatur.jpg] <http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/SBA_Badges_2018_mailsignatur.jpg>

meltdown03, 2020-05-30:
The very next one failed.

-9
LWMQTT_MISSING_OR_WRONG_PACKET
6
LWMQTT_UNKNOWN_RETURN_CODE

Maybe it was my internet connection and you'll have better luck.

AkeJohansson, 2020-05-30:
So far so good I have done 15 refreshing and no not connected yet.
It seems that this solved the problem for me.
Cross fingers.🙂

Med vänliga hälsningar

Åke Johansson
ake.h.johansson@bravida.se
Tekniker / Technician

Bravida Sverige AB

[http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Logga.jpg]<http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Logga.jpg>
Telefon +46730-477647

Vx. +46470-377 00

Hjalmar Petris väg 32, Växjö

Besök oss gärna på www.bravida.se<https://www.bravida.se/>
Visit us at www.bravida.se/en/<https://www.bravida.se/en/>
Facebook.com/bravida.se<http://www.facebook.com/bravida.se>

[http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/Bravida_mailbanner.jpg]<http://brain.corp.bravida.com/Global/Sverige/Aktuellt/Bilder/2016/Bravida_mailbanner%20%20liten%20.jpg>
 [http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/SBA_Badges_2018_mailsignatur.jpg] <http://brain.corp.bravida.com/Global/Sverige/Info-marknad/Bilder/SBA_Badges_2018_mailsignatur.jpg>

meltdown03, 2020-05-30:
That's good, the one error I had showed up on the Google Cloud log as
"ALREADY_CONNECTED". Maybe the 50ms delay is a little too short. But one
error once in a while doesn't bother me that much.

meltdown03, 2020-05-30:
Scratch that, it was the "FAILED_PRECONDITION" error 9

raphael-bmec-co, 2020-05-31:
I'm back on this today. Just a heads up that I have a lot of issues with the CloudIoTMqtt abstraction layer: #161, #166, #167 and #169 ([here](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/169#issuecomment-635248347)). I have now seen that the JWT refresh calls the synchronous (instead of the async) connect method and my architecture requires that the code doesn't get stuck in that loop. One of the reasons why is that it jumps to 60s reconnect delay instead of implementing the google specified exponential backoff.

I'll let you know if it solves the issues. 

Is anyone still having issues?

AkeJohansson, 2020-05-31:
Ok I set mqttConnect(false); in CloudIoTCoreMqtt.cpp it's better but not perfect. I have done 100 refresh JWT every 5 minutes and only 9 resulted in Settings incorrect or missing a cyper for SSL so it's clearly an improvement.

Google IoT cloud claim JWT refreshing every hour so I set the jwt_exp_secs = 40*60 seconds allow 20 minutes for drift and Google server is waiting 10 more minutes before JWT  expires. So I think this should work for me.

Med vänliga hälsningar

Åke Johansson
ake.h.johansson@bravida.se
Tekniker / Technician

Bravida Sverige AB

raphael-bmec-co, 2020-05-31:
Hi.

In my opinion, there was a lot being done wrong int CloudIoTCoreMqtt. I think I've made some major improvements in the attached including:
1. Checks done in lower level libraries were being repeated.
2. Flags were being set in methods called log...
3. Changed to Arduino ESP logs.
4. Introduced a state machine.
5. Done away with calls to global functions and created an instantiable class. 
6. Implemented primary and backup certs.
7. Functional exponential backoff as per google quotas. 
8. Implemented state callbacks to the instantiating methods.

Please feel free to adapt as needed for you own use (wifi connection is external and there are some dependances that you will need to code out - specifically it id designed for BLE to set private key and identity).

As far as I can tell there are three core issues.

1. Google seems to indicate that the JWT [can be refreshed without stopping the connection](https://cloud.google.com/iot/docs/how-tos/credentials/jwts#refreshing_jwts).  However, when i call connect with `skip = true` the connect fails and the gcp logs show that the device unexpectedly disconnected. Moreover, the actual code they include clearly disconnects before reconnecting with the new JWT and other documentation refers to reconnecting. Based on this, I am concluding that disconnect must be called before connect.

2. If connect is called with `skip = true` the connection is broken and the gcp log shows an error. However, if `mqtt_client_->disconnect();` is called first then the disconnection is not an error. Based on this I am concluding that the connection must be closed with `mqtt_client_->disconnect();` before calling connect with `skip = false`.

3. When the connection fails, the exponential backoff jumps to a 60s delay. But retying as per Google's exponential backoff spec will actually succeed resulting in less than ~2s downtime. 

Now that I properly close and open the connection i no longer get and errors on the gcp side.

However, the refresh still periodically fails with error:
```
[W][co_bmec_ciotc.cpp:409] connect(): mqtt_client_ last error: -10 -LWMQTT_CONNECTION_DENIED
[W][co_bmec_ciotc.cpp:412] connect(): mqtt_client_ return code: -10 - LWMQTT_NOT_AUTHORIZED
```

This is returned by the mqtt connack packet but the bounce does not reflect on the GCP logs at all. Which leads me to believe it may be related to the 5 second JWT refresh and hitting the quotas. 

I am now running a test with 90 refresh and will report back on the results. So far I have done 8 refreshes with no issue.

Hope this helps!

[co_bmec_ciotc.zip](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/files/4707605/co_bmec_ciotc.zip)

raphael-bmec-co, 2020-05-31:
Okay after many more hours, I'm calling the failure rate acceptable based on being able to reconnect within a second and no errors being thrown in CGP. Optimistically looking forward to someone else squashing this one.

raphael-bmec-co, 2020-05-31:
Wait... There is more. I have ruled out the created and expired times in the JWT as the issues as this gives a different error. However, I can force the same error everytime by passing an incorrect project id into the JWT creation. So I do believe this is an issue with the JWT. Will see if I can pin it down.

raphael-bmec-co, 2020-05-31:
### Guys. I've done it. I've found the bug!!!

By hard coding changes to the connection parameters I worked out that the issue happened if the JWT was wrong in anyway other than the start and end dates being wrong. I knew my project name was correct because it sometimes connected so I started looking at the JWT generation.

The JWT is generated in `jwt.cpp`.

I saw this line: `String sha256 = get_sha(header_payload_base64);` and the alarm bells went off. Why is 64 encoded data being passed around in strings and sure enough:

```
[E][jwt.cpp:142] CreateJwt(): header: {"alg":"ES256","typ":"JWT"}
[E][jwt.cpp:143] CreateJwt(): payload: {"iat":1590956636,"exp":1590956641,"aud":"co-bmec-arc005-prod"}
[E][jwt.cpp:146] CreateJwt(): header_payload_base64: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1OTA5NTY2MzYsImV4cCI6MTU5MDk1NjY0MSwiYXVkIjoiY28tYm1lYy1hcmMwMDUtcHJvZCJ9
raw: 60:8A:E0:AA:2:F6:B9:E1:FB:32:AB:4A:FE:6D:C2:DD:0:C0:49:39:21:FB:E4:6D:3:B8:F0:7F:F:9D:1D:3D:
String: raw: 60:8A:E0:AA:2:F6:B9:E1:FB:32:AB:4A:FE:6D:C2:DD:
[E][jwt.cpp:148] CreateJwt(): sha256: `Šàªö¹áû2«JþmÂÝ
[I][co_bmec_ciotc.cpp:467] get_jwt(): JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1OTA5NTY2MzYsImV4cCI6MTU5MDk1NjY0MSwiYXVkIjoiY28tYm1lYy1hcmMwMDUtcHJvZCJ9.wKBpoMipAnalek4H6PzmiQ26BwuQl0WpGskvkF9ION4J2b6zxBvQMOr6pLNtgTsBjo7gtLpqbGZeciUwQNxTJQ==
[W][co_bmec_ciotc.cpp:413] connect(): Attempt 1 failed to connect ciotc. Trying again after 1114ms
[W][co_bmec_ciotc.cpp:414] connect(): Host: mqtt.2030.ltsapis.goog:443
[W][co_bmec_ciotc.cpp:415] connect(): ClientId: projects/co-bmec-arc005-prod/locations/europe-west1/registries/arc003/devices/arc003_SN0001
[W][co_bmec_ciotc.cpp:416] connect(): JWT: eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1OTA5NTY2MzYsImV4cCI6MTU5MDk1NjY0MSwiYXVkIjoiY28tYm1lYy1hcmMwMDUtcHJvZCJ9.wKBpoMipAnalek4H6PzmiQ26BwuQl0WpGskvkF9ION4J2b6zxBvQMOr6pLNtgTsBjo7gtLpqbGZeciUwQNxTJQ==
[W][co_bmec_ciotc.cpp:417] connect(): mqtt_client_ last error: -10 - LWMQTT_CONNECTION_DENIED
[W][co_bmec_ciotc.cpp:420] connect(): mqtt_client_ return code: -10 - LWMQTT_NOT_AUTHORIZED
```
 When the generated sha256 contains the termination caracter it gets terminated and BOOM no connection. 

@gguuss I have tried (hard) to find a fix but the code just crashes with all the casting across data types. Please help.

meltdown03, 2020-05-31:
When I was debugging this with my FTDI, I noticed that VSCode debugging was
cutting my JWT keys off by a few characters in the connect() function. I
thought it was just showing me the first couple hundred bytes, so I didn't
mention it. Haha. Glad that I know I can count on it to be accurate. And
it's fixed.

meltdown03, 2020-05-31:
Maybe not fixed exactly.... I was trying to get rid of all the Strings but they were everywhere. I guess I was actually missing more than the null-termination char, i was missing about 10 chars, so maybe it was just showing me the first 200 bytes

meltdown03, 2020-05-31:
Is there a way to use the C JWT implementation to get rid of all the Strings: https://github.com/GoogleCloudPlatform/iot-device-sdk-embedded-c/blob/master/src/libiotc/iotc_jwt.c

raphael-bmec-co, 2020-06-01:
I reached that point too. But that library has a lot of dependencies and I was having issues with linking and I'm not actually clear on how to mix c into c++ so I'm sticking with my work around until the issue is resolved here.

raphael-bmec-co, 2020-06-01:
Okay. After running all night, with a refresh rate of 5s (over 4000 reconnects) there were 623 occurrences of this error and 25 other unknown failures to connect (the log have been overwritten so no idea what these were due to). I think we can safely say this is the core issue here with a failure rate of 0.625% probably being acceptable.

mars000, 2020-08-08:
Is there a solution to original issue ? I'm finding the same error, how do I solve it 
```
connecting...Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883

```

gguuss, 2020-10-15:
@mars000 This is generally due to having an invalid parameter in ciotc_config.h, there are additional details on correctly configuring the examples in [this Issue](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/92).

closed 2020-10-15 by gguuss

CHeffernan087, 2021-06-07:
Posting a link to this thread as it helped me solve this error 👀 

[https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/221](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/221)

For me it was a problem with the CA cert and the wifi secure client

aniketdhole07, 2022-02-18:
I had same issue and solved it by changing the key from root.pem file to getting key from google cloud openssl command. 

```openssl s_client -showcerts -connect mqtt.2030.ltsapis.goog:8883```

And copy the first key in ciotc_config.h

The error was because the keys from root.pem work on `mqtt.googleapis.com:8883` and not on `mqtt.2030.ltsapis.goog:8883` url.