GoogleCloudPlatform/google-cloud-iot-arduino #146: jwt refresh crashing

sphonala, 2020-01-18:
connecting...Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/kwikswitch-a9be0/locations/us-central1/registries/mmtestreg/devices/k11b701
Waiting 60 seconds, retry will likely fail
Refreshing JWT

Exception (28):
epc1=0x4022bdda epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffce0 end: 3fffffc0 offset: 01a0
3ffffe80:  00000000 ff000000 3ffffecc 4021c482  
3ffffe90:  3ffffecc 3ffffdf8 3ffffecc 00000e10  
3ffffea0:  3fff334c 3fff3384 3fff3384 4021c5b2  
3ffffeb0:  3fff334c 3fff3384 3fffff10 40211797  
3ffffec0:  00000000 00000000 ffff1808 3fff36f4  
3ffffed0:  0010001f ffffff4c 3fffff10 40210ca5  
3ffffee0:  5e22a992 00000000 3fff1808 4021b400  
3ffffef0:  4021a894 3fffff4c 3fff1808 00000000  
3fffff00:  3fff17e8 3fff0d7c 3fffff40 402026bd  
3fffff10:  00000000 00000000 ff21002f 00000000  
3fffff20:  00000000 ff00002f 0000ea60 00000000  
3fffff30:  3fff1808 3fffff4c 3fff501c 4021206d  
3fffff40:  00000000 00000000 ffff1808 3fff521c  
3fffff50:  0054005f ffff1808 3fff1808 40210ca5  
3fffff60:  3fff521c 3fff5114 3fff1808 40210ca5  
3fffff70:  3ffe9152 3fff0540 3fff1808 3fff1938  
3fffff80:  3fffdad0 3fff1808 3fff0d98 40202776  
3fffff90:  3fffdad0 3fff1808 3fff0d98 4020722a  
3fffffa0:  3fffdad0 00000000 3fff1908 4021d660  
3fffffb0:  feefeffe feefeffe 3ffe85c8 40100a9d  
<<<stack<<<

last failed alloc call: 4021C2C4(208)

fabltd, 2020-02-03:
Bump

I am seeing the same:

Once it fails there is no recovery. Other than a hardware reset. Help...

-----

Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: xxxxxxxxxxxxxxxxxx
Waiting 60 seconds, retry will likely fail
Refreshing JWT
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-17040) RSA - The public key operation failed : BIGNUM - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -17040
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: xxxxxxxxxxxxxxxxxxxxxxxx
Waiting 60 seconds, retry will likely fail
Refreshing JWT
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-17040) RSA - The public key operation failed : BIGNUM - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -17040
not connected

gguuss, 2020-02-06:
I have seen this before and the issue went away when I tried switching the library version or device. Try copy/pasting the stackdump into the [ESP exception decoder tool](https://github.com/me-no-dev/EspExceptionDecoder) to see where the crash is occurring in code and let us know along with the version of the library you're usoing. I'm guessing this is occurring on an ESP8266, if you have the version of the ESP8266 SDK and your board type, it may help with reproducing the issue.

vmehmeri, 2020-04-08:
I am having a similar issue with Arduino MKR1000. The reason seems to be that the token refresh is done against the URL mqtt.2030.ltsapis.goog:8883, which I wasn't able to upload a certificate for because using the SSL certificate upload tool for MKR1000 with the domain name mqtt.2030.ltsapis.goog:8883 (or :443) throws an error saying that the certificate must have a public RSA key. The MKR1000 does not have a data upload tool as far as I'm aware so I can't manually upload the file, so now I'm stuck...

gguuss, 2020-10-15:
Please reopen if you're continuing to have the issue, this may be resolved in newer versions of the library and through updating the stack size for your board as specified in the README.

closed 2020-10-15 by gguuss