GoogleCloudPlatform/google-cloud-iot-arduino #147: Error compiling for board esp 32 wrover module

Gitanjali-apet, 2020-01-22:
I'm using the camera example and un that I'm getting error while compiling for selected board. Tried using different boards

gguuss, 2020-02-06:
Each of the camera modules requires some customization for [this configuration setting](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/complex/esp32/camera/camera.ino#L62-L117) in the demo application. I believe I have gotten the WROVER module working before but there is still some variance from board to board, such as having PSRAM on-board. Check the data sheet for the specific WROVER you have and if you have a fix, I'd love to review a PR for the board settings and an additional board definition in that block.

gguuss, 2020-03-24:
Oof, there's also a bug in the code (sorry!). [Fixed in this PR](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/pull/153).

Add the following code to `ciotc_config.h`:
```
// Time (seconds) before expiring JWT token.
const int jwt_exp_secs = 60*20;
```

gguuss, 2020-10-15:
I believe this was due to the bug that we fixed, the CI should also help to prevent these breakages.

closed 2020-10-15 by gguuss